mainsail-crew/mainsail #1507: Choose image background color based on color of the thumbnail content

EmJay276, 2023-08-04:
### Requested feature:

If I print something in black, and use the dark theme, it is really hard to see the object in the thumbnail.
Is it possible to choose the background color of this black based on the color of the picture?

If you point out to me where it could be (possibly) done in the code, I can have a look at it.

[image]

meteyou, 2023-08-04:
sry, but i would recommend to use a other color in your slicer or use a different background color in your slicer. mainsail just display the thumbnail as it is...

EmJay276, 2023-08-05:
It is a PNG with a transparent background, that's why I think mainsail could easily handle this by adjusting the background color of the widget based on the image color.

ja2ui0, 2023-08-16:
EmJay276 is correct, the slicer outputs a transparent PNG for the thumbnail and therefore the background will be whatever Mainsail chooses for it to be. It should select lighter or darker colors within a range depending on the lightness or darkness of the thumbnail, or at least allow the user to override the default color (or brightness at least, perhaps with a slider).

EmJay276, 2023-08-26:
I figured out what needs to be changed

`<div class="v-image__image v-image__image--cover" style="background-image: url(&quot;http://voronika/server/files/gcodes/.thumbs/VC_sturdy_handles-400x400.png?timestamp=1693058858.16708&quot;); background-position: center center;"></div>`

needs an additional `background-color: #FFF;` (or whatever color)

Before
[image]

After
[image]

dw-0, 2023-08-27:
Don't use a full black for the color. I just tested it with dark colors and with an rgb value of (40,40,40) its imho still fine.
[image]
hard to see, but with your black color you don't even get those natural shadows. so thats even an issue of the slicer itself, its just a flat black area you get there. imho it looks horrible even with a white or light background.

ja2ui0, 2023-08-30:
I don't use full black for the color anyway, I use a dark grey in the slicer so it's more visible in the slicer itself. It's great that Cura allows me to change filament colors to improve the viewing experience. Like OP, I'd still recommend an option in the mainsail interface to make the thumbnail background more contrasted.

meteyou, 2023-08-30:
@ja2ui0 i already implemented it -> https://github.com/mainsail-crew/mainsail/pull/1535

PR #1535 (2023-10-07): feat: add optional background color for big gcode thumbnails
## Description

This PR adds a optional background color to gcode thumbnails.

## Related Tickets & Documents

fixes: #1507 

## Mobile & Desktop Screenshots/Recordings

Start print dialog:
[image]

Status panel:
[image]

Gcode list:
[image]

Settings:
[image]

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2023-10-07 by meteyou