mainsail-crew/mainsail #1508: Automatically connect to first online printer

roobre, 2023-08-04:
### Requested feature:

When configured to use the json-defined instance list, and even if there is only one instance on the list, mainsail still asks every page load to which printer I want to connect:

[image]

I think it would be great if we had a config option which, if enabled, told mainsail to automatically connect to a printer if available.

### Solves the following problem:

It would reduce the number of clicks I have to make in order to view, or intervene the printing process. This is particularly annoying in mobile, where it is common for browsers to kill inactive tabs.

meteyou, 2023-08-27:
just don use the json DB. just add your ip and port above from it and use moonrakerDB. then it will connect to this printer.

closed 2023-08-27 by meteyou

roobre, 2023-08-28:
Hey @meteyou , thanks for your suggestion! I'm trying to understand a bit better how this would work on my setup. Where is this moonrakerDB stored? Would that entry be saved if I access Mainsail from different devices, or would I need to configure the printer on each browser I access Mainsail from?

meteyou, 2023-08-28:
@roobre only printers, which you added in the mainsail settings, will be stored in the Moonraker DB. The Moonraker DB is the Moonraker instance which you connect. And you change the config.json, which is the same file for every device.