mainsail-crew/mainsail #1510: gcode hints

Matszwe02, 2023-08-05:
### Requested feature:

Add gcode hints to the terminal. There's `gcodes.md` file inside printer_data. Make this hint tool search for gcode section and display it as a hint for gcode. It can also be added as a description to when I click `?` and see list of gcodes.

### Solves the following problem:

I don't know what exactly klipper's gcodes naming is and what parameters do they get. Creating hints makes it obvious that I need to type `x` parameter to achieve `something` or to use `SET_HEATER_TEMPERATURE` with parameter `HEATER`.

### Additional information:

Every `h4` section is titled by the gcode macro name. The `h3` prior to it describes which module is needed to be loaded in configuration file (look at it to check if display hint or not?).

meteyou, 2023-08-27:
just my 2 cents... it would be a better solution to fix the gcode descriptions in Klipper, which are in in the `?` dialog. These are normally available via API.

Matszwe02, 2023-08-27:
> just my 2 cents... it would be a better solution to fix the gcode descriptions in Klipper, which are in in the `?` dialog. These are normally available via API.

So you suggest changing how klipper handles the gcode examples? Too bad klipper doesn't have feature request section...

meteyou, 2023-08-29:
- Klipper has all these texts directly in the code: https://github.com/Klipper3d/klipper/blob/master/klippy/extras/bed_mesh.py#L244
- Mainsail grab them via Moonraker: https://moonraker.readthedocs.io/en/latest/web_api/#get-gcode-help

These are all the data behind the `?` dialog in Mainsail. I would love it when Klipper provided all these texts here.

Matszwe02, 2023-08-29:
> * Klipper has all these texts directly in the code: https://github.com/Klipper3d/klipper/blob/master/klippy/extras/bed_mesh.py#L244
> * Mainsail grab them via Moonraker: [...]

ok, so they are hardcoded. I think that mainsail could pull them from the gcode reference instead. I don't know at which point it could be better to inject the better hints, the mainsail or klipper. I think mainsail has access to these files and it could pretty easily parse them, even hint which parameters are needed or optional for each command as you type. In my opinion, doing it in klipper, it would need to rewrite how this gcode is sent over moonraker, by reading the gcodes.md file and splitting it into gcodes. However it would improve writing and modifying features, because these descriptions are stored only in one place.
I think the better option is to do it by mainsail - simpler to write, and it could be used as 'extended' gcode hints side by side with the standard one, maybe with option to display it like code editors display hints when typing.

meteyou, 2023-08-29:
I have a completely different opinion. Mainsail is only running in your browser and have no part of it on your host. so for example, mainsail have to repull and parse this file every time, when you open the browser tab. next thing is, why have klipper 2 different descriptions for the gcodes? klipper could also use this description to auto generate there docs. this is the right way in my opinion.

Matszwe02, 2023-08-29:
> I have a completely different opinion. Mainsail is only running in your browser and have no part of it on your host. so for example, mainsail have to repull and parse this file every time, when you [...]

Right, good point. So I think this FR could be forwarded to klipper's discourse group

meteyou, 2023-08-29:
yes. this would be a good idea.