mainsail-crew/mainsail #1520: Heightmap display has a fencepost error in coordinate calculations

dfinberg, 2023-08-15:
### Mainsail Version:

2.7.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

In looking at the heightmap mesh, it displays the coordinate points of the x/y points.  my bed is configured with a min of 10, max of 270 in the Y direction, 9x9. The highest point is at the left rear, at 10,270.  In the right hand panel the max point is instead showed as "Max [10, 241.1]". 241.1 = 10+(270-10)*8/9 instead of the correct divisor of 8.

### What did you expect to happen instead?

coordinate should be calculated correctly.

### How to reproduce this bug?

display a heightmap and look at the max/min coordinates on the right panel, assuming they are not at the first point probed.

dfinberg, 2023-08-16:
I've never used vue before, but something like 
```
--- a/src/pages/Heightmap.vue
+++ b/src/pages/Heightmap.vue
@@ -985,17 +985,23 @@ export default class PageHeightmap extends Mixins(BaseMixin, ControlMixin) {
             }
         })
 
+       let x_length = this.bed_mesh.probed_matrix[0].length - 1;
+       if (x_length < 1) x_length = 1;
+
+       let y_length = this.bed_mesh.probed_matrix.length - 1;
+       if (y_length < 1) y_length = 1;
+
         const positionX =
             Math.round(
                 (this.bed_mesh.mesh_min[0] +
-                    ((this.bed_mesh.mesh_max[0] - this.bed_mesh.mesh_min[0]) / this.bed_mesh.probed_matrix[0].length) *
+                    ((this.bed_mesh.mesh_max[0] - this.bed_mesh.mesh_min[0]) / x_length) *
                         (col - 1)) *
                     10
             ) / 10
         const positionY =
             Math.round(
                 (this.bed_mesh.mesh_min[1] +
-                    ((this.bed_mesh.mesh_max[1] - this.bed_mesh.mesh_min[1]) / this.bed_mesh.probed_matrix.length) *
+                    ((this.bed_mesh.mesh_max[1] - this.bed_mesh.mesh_min[1]) / y_length) *
                         (row - 1)) *
                     10
             ) / 10
@@ -1023,17 +1029,23 @@ export default class PageHeightmap extends Mixins(BaseMixin, ControlMixin) {
             }
         })
 
+       let x_length = this.bed_mesh.probed_matrix[0].length - 1;
+       if (x_length < 1) x_length = 1;
+
+       let y_length = this.bed_mesh.probed_matrix.length - 1;
+       if (y_length < 1) y_length = 1;
+
         const positionX =
             Math.round(
                 (this.bed_mesh.mesh_min[0] +
-                    ((this.bed_mesh.mesh_max[0] - this.bed_mesh.mesh_min[0]) / this.bed_mesh.probed_matrix[0].length) *
+                    ((this.bed_mesh.mesh_max[0] - this.bed_mesh.mesh_min[0]) / x_length) *
                         (col - 1)) *
                     10
             ) / 10
         const positionY =
             Math.round(
                 (this.bed_mesh.mesh_min[1] +
-                    ((this.bed_mesh.mesh_max[1] - this.bed_mesh.mesh_min[1]) / this.bed_mesh.probed_matrix.length) *
+                    ((this.bed_mesh.mesh_max[1] - this.bed_mesh.mesh_min[1]) / y_length) *
                         (row - 1)) *
                     10
             ) / 10

```

meteyou, 2023-08-27:
Thx for reporting this bug and your solution!

PR #1533 (2023-09-04): fix: fix min/max positions in heightmap current mesh data panel 

## Description

This PR fix the min/max positions in the heightmap page in the "current mesh" panel. The calculation of the position from the datapoints was incorrect.

## Related Tickets & Documents

fixes #1520 

## Mobile & Desktop Screenshots/Recordings

[image]

## [optional] Are there any post-deployment tasks we need to perform?
closed 2023-09-04 by meteyou