mainsail-crew/mainsail #1527: output_pin entry missing from dashboard

HitLuca, 2023-08-23:
### Mainsail Version:

v2.7.1

### Browser:

Firefox

### Device:

Laptop

### Operating System:

macOS

### What happened?

Looks like the entry for a beeper isn't showing up as a toggle on the main Mainsail interface (under the miscellaneous card). It was there before but I'm not sure when this issue started happening.

This is my beeper configuration in `printer.cfg`:

```yaml
[output_pin beeper]
pin: EXP1_1
pwm: True
value: 0
shutdown_value: 0
cycle_time: 0.001
scale: 1000
```

### What did you expect to happen instead?

Have the beeper toggle show up in the mainsail ui under the `Miscellaneous` card

### How to reproduce this bug?

create an `output_pin` entry in printer.cfg and check the dashboard

### Additional information:

Other entries are shown in my `Miscellaneous` card, but none of them are `output_pin` instances

[image]

meteyou, 2023-08-23:
every `output_pin` with `pwm: true` will be displayed as a slider. a beeper can create different sounds, that why you need pwm. so this is no bug and when you need more help to setup your printer, pls use our discord.

closed 2023-08-23 by github-actions[bot]

HitLuca, 2023-08-23:
The beeper isn't displayed at all, neither as a toggle nor as a slider

meteyou, 2023-08-23:
@HitLuca pls join our discord server and open a ticket in the support-forum...