mainsail-crew/mainsail #1539: Chrome browser notification pop-ups (first layer done, start print, print end, error etc.)

PhilBaz, 2023-08-28:
### Requested feature:

Chrome browser notification pop-ups (first layer done, start print, print end, error etc.)

Would be great to have these types of notifications built right into mainsail.

### Solves the following problem:

Dont know whats happening?

### Additional information:

Mainsail is the best!

meteyou, 2023-08-28:
Browser notification is only possible with Https/SSL. This is not tha case in 99.9% of all installations. The next issue is, that mainsail is only running in your browser. There is no service on the host, so this would also only be possible when you have a browser tab open with your mainsail instance.

So thanks for your request, but there are too many issues for this feature. I would recommend your to use the Moonraker notifications. Moonraker is running all the time on your pi/host.

Maybe this is possible in the future, but right now the SSL certificate is the biggest problem for this feature.

closed 2023-08-28 by meteyou

PhilBaz, 2023-08-28:
@meteyou Understood. Thanks for considering it.