mainsail-crew/mainsail #1540: Option to keep Z Thermal Adust units in mm, and not go to microns.

PhilBaz, 2023-08-28:
### Requested feature:

Please add an option to keep Z Thermal Adjust units in mm, and not go to microns. 

It would be nice to have a consistent scale for Z Thermal Adjust units. I think most people are accustomed to thinking in mm. And somehow it's annoying to have to remember what decimal place a micron is, and remind yourself what it means. 

Ohhh, please add this option.

### Solves the following problem:

I'm not that smart?

### Additional information:

Thank you!

meteyou, 2023-08-28:
if more people come forward here who see it as a issue, then we will change it. currently there is no need / not enough interest to change it.

meteyou, 2023-10-29:
I will close this FR, because there are no more inputs / new users for this...

closed 2023-10-29 by meteyou

PhilBaz, 2023-10-30:
Ahh, shit... I should have been campaigning on discord. F!