mainsail-crew/mainsail #1549: Unusable Confirmation popup on Emergenct Stop

pfgh1, 2023-09-07:
### Mainsail Version:

2.7.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When clicking on emergency stop a popup window launches asking for confirmation. However the action (emergency stop/firmware shutdown,) immediately occurs with the popup serving no function.

### What did you expect to happen instead?

Either no prompt like in previous versions or have the prompt actually stall the operation until yes/no is clicked.

Really the best would probably be to add a configuration option to enable/disable the confirmation. Some people might find value in protection from accidentally clicking emergency stop whereas others might want emergency stop to be just that... immediate.

Thanks for the great work!

### How to reproduce this bug?

Click Emergency Stop during a print job.

meteyou, 2023-09-07:
this is already fixed -> https://github.com/mainsail-crew/mainsail/pull/1526

closed 2023-09-07 by meteyou

pfgh1, 2023-09-07:
Thanks but I am on 2.7.1 and still get the popup?

meteyou, 2023-09-07:
Yes, because I fixed the issue after V2.7.1

pfgh1, 2023-09-07:
d0h sorry. Thanks again for the awesome work on this project.