mainsail-crew/mainsail #1553: Use Weblate to facilitate translation

Daviteusz, 2023-09-08:
### Requested feature:

I translate Mainsail using Weblate because it is a very convenient program for translators and is overall perfect for this project.

When the translation is completed, e.g. every 12 hours, the changes are automatically sent as a Pull Request. The entire operation rests on the shoulders of the weblate bot, translators don't need to know git usage to add new translations or update them.

### Additional information:

[image]
[image]
[image]
[image]
[image]
[image]

meteyou, 2023-11-20:
@Daviteusz can I ask, why you close this FR? I hadn't time until now to look closer at this tool.

Daviteusz, 2023-11-20:
There was no response from almost three months, so I assumed there was no interest 😅

meteyou, 2023-11-20:
sorry, that was my mistake. i should have written a short comment to let you know that i need some time to have a look at this tool. unfortunately, i don't have any experience with it yet.

juliazero, 2025-02-21:
I want to update translations but...
https://hosted.weblate.org/projects/mainsail/mainsail/#alerts
```
Weblate could not merge upstream changes while updating the repository.

Auto-merging src/locales/es.json
Auto-merging src/locales/hu.json
Auto-merging src/locales/nl.json
CONFLICT (content): Merge conflict in src/locales/nl.json
Auto-merging src/locales/tr.json
Auto-merging src/locales/zh_TW.json
CONFLICT (content): Merge conflict in src/locales/zh_TW.json
Automatic merge failed; fix conflicts and then commit the result.
 (1)

Appeared 3 months ago, last seen 11 hours ago
```
To fix this:

1. https://hosted.weblate.org/projects/Mainsail/Mainsail/#repository
2. Click first red reset button in danger zone section :)

meteyou, 2025-04-22:
@juliazero sry for the long long long delay! I fixed the issue and it should be possible to update translations now.

closed 2025-04-22 by juliazero