mainsail-crew/mainsail #1559: Webcam Support - mediamtx - WHEP?

hostrup, 2023-09-19:
### Requested feature:

Hi! 
I am trying to configure mainsail to use my unifi g3 intant cam. 
I was unable to use the rtsp or rtsps stream directly from unifi protect. 
so instead i choose to install mediamtx - as i saw mainsail supported that. 

Now mediamtx is wokring - and i have access to my unifi cams from my mediamtx server. 

i am however not able to use it inside mainsail. I keep getting errors - becuase it's trying to start a WS session. 
as far as i understand - from this link: 

https://github.com/bluenviron/mediamtx/discussions/1845

mediamtx - is not supporting websockets anymore - instead they suggest to use WHEP. 

I found this PR: 
https://github.com/mainsail-crew/mainsail/pull/1558

Can anyone tell me if this will actually solve the problem with never version of mediamtx - and provide the ability to use webrtc via WHEP inside mainsail?

meteyou, 2023-09-19:
Yes. the PR #1558 is the WHEP implementation for this webcam clientl

hostrup, 2023-09-19:
awesome! 
I will be ready to test it - onced it's released! 
Thanks alot :D 

br Ronni

PR #1558 (2023-10-07): refactor: update webcam "WebRTC MediaMTX" client
## Description

This PR update the "WebRTC MediaMTX" client for MediaMTX v0.23 and above.

## Related Tickets & Documents

fixes: #1559

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?
closed 2023-09-19 by hostrup