mainsail-crew/mainsail #1572: Heightmap not showing nothing in browser

RudrigoMaciel, 2023-09-28:
### Mainsail Version:

last one, juste updated

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

After update via KIAUH, the button HEIGGHTMAP in mainsail web interface doesn´t showing nothing. 
BED_MESH_PROFILE LOAD=default it's. 
I am using a Kackender probe and adptative meshing, that was work ok before reinstall klipper, Moonraker  and Mainsail (all config file was prepreviously saved. 
Everythin is working ok, only HEIGGHTMAP  doesn't show nothing (it means: not even the browser tab is loaded.). It's no reaction when I click in HEIGGHTMAP  button.

### What did you expect to happen instead?

When we click em HEIGGHTMAP  button, something happens, but nothing happens..

### How to reproduce this bug?

just click the HEIGGHTMAP button in the browser

RudrigoMaciel, 2023-09-28:
It was a browser cache problem. 
Just clean the cache after hours..

closed 2023-09-28 by RudrigoMaciel

soleshoe, 2024-06-07:
i had the same error, and adding a [bed_mesh] section in printer.cfg resolved it

brownem, 2025-02-22:
> It was a browser cache problem. Just clean the cache after hours..

Confirming the solution; clearing the cache resolved it for me as well. I already had [bed_mesh] in my printer.cfg, and it seemed to be working correctly as far as the mechanics were concerned. But clicking on "Heightmap" in the side bar produced zero reaction from the software. Using Firefox, clearing the cache and cookies for all sites brought up the heightmap once I refreshed the mainsail page for my printer.