mainsail-crew/mainsail #1579: Allow embedding PrettyGCode (pgcode) somewhere within MainSail's UI

Ybalrid, 2023-10-03:
### Requested feature:

There's a cool little klipper addon called [Pretty GCODE Viewer](https://github.com/Kragrathea/pgcode) (that happens to be installable via [KIAUH](https://github.com/dw-0/kiauh))

It allows visualizing the current print in "semi real time" in a 3d view that respond better than the embedded gcode viewer of mainsail.

The way this application works is by running it's own web server on a non standard port on localhost.

I think it would be neat to allow embedding that web page as a different option for the G-Code viewer 🙂

meteyou, 2023-10-07:
thanks for your feature request. but i would decide against it from the current point of view, because the 3d gcode viewer is currently being massively rebuilt by the maintainer (@Sindarius) and as far as i have tested it, the next version will be a lot faster and lighter.

closed 2023-10-07 by meteyou

Ybalrid, 2023-10-07:
> thanks for your feature request. but i would decide against it from the current point of view, because the 3d gcode viewer is currently being massively rebuilt by the maintainer (@Sindarius) and as far as i have tested it, the next version will be a lot faster and lighter.

Interesting, thank you! Looking forward to see a better performing gcode viewer in Mainsail!

Sindarius, 2023-10-07:
You can check out the current test version at http://sindarius.com/v2

It is not complete but will give you an idea of where it is going.

speendo, 2024-03-17:
is there an update on v2?
The link that @Sindarius posted is not working anymore :(

Sindarius, 2024-03-17:
It is still a work in progress. It's something that I work on as I get time between other projects.

meteyou, 2024-07-18:
@swatkxs sry, but did you read the answer?