mainsail-crew/mainsail #1582: RESUME - Option to move to XY first then lower to Z

PhilBaz, 2023-10-05:
### Requested feature:

The current implementation of RESUME (along with park_pause_spot) moves the tool head diagonally (in Z) to the XYZ position where the toolhead will resume. 

This caused a nozzle catch and layer shift for me once or twice.

I think it would be ideal to at least have an option to move first to XY position then down to Z to minimize the chance of any nozzle catch. the implementation could have its own Z Height parameter (above the resume position) or use the Z lift height implemented for pause.

Just as it makes sense to lift on pause it makes sense to unlift on resume. 

### Solves the following problem:

Nozzle catch/crash/layershift when hitting a curled edge or something else.

### Additional information:

Thank you for everything you do!

meteyou, 2023-10-05:
This have nothing Todo with mainsail. Mainsail just send the macro/gcode name. The rest is your Klipper config.

The position can be set in/with your mainsail.cfg, but resume is mostly a Klipper thing.

closed 2023-10-05 by meteyou

PhilBaz, 2023-10-05:
@meteyou 

It has as much to do with mainsail as the 'variable_custom_park_dz' Z lift parameter in the 'Client variable' control panel for the mainsail.cfg, no?

It has about the same amount as most of the functions in the 'Client variable' macro, no?

meteyou, 2023-10-05:
Both have nothing Todo with mainsail as Webinterface. It only send "PAUSE" or "RESUME" via Moonraker to Klipper. Nothing more.

PhilBaz, 2023-10-05:
@meteyou 

Do you happen to know who establishes the _TOOLHEAD_PARK_PAUSE_CANCEL behaviour? Maybe I should get in touch with them?

##### internal use #####
[gcode_macro _TOOLHEAD_PARK_PAUSE_CANCEL]
description: Helper: park toolhead used in PAUSE and CANCEL_PRINT
gcode:
  ##### get user parameters or use default #####
  {% set client = printer['gcode_macro _CLIENT_VARIABLE']|default({}) %}
  {% set velocity = printer.configfile.settings.pause_resume.recover_velocity %}
  {% set use_custom     = client.use_custom_pos|default(false)|lower == 'true' %}
  {% set custom_park_x  = client.custom_park_x|default(0.0) %}
  {% set custom_park_y  = client.custom_park_y|default(0.0) %}
  {% set park_dz        = client.custom_park_dz|default(2.0)|abs %}
  {% set sp_hop         = client.speed_hop|default(15) * 60 %}
  {% set sp_move        = client.speed_move|default(velocity) * 60 %}
  ##### get config and toolhead values #####
  {% set origin    = printer.gcode_move.homing_origin %}
  {% set act       = printer.gcode_move.gcode_position %}
  {% set max       = printer.toolhead.axis_maximum %}
  {% set cone      = printer.toolhead.cone_start_z|default(max.z) %} ; height as long the toolhead can reach max and min of an delta
  {% set round_bed = True if printer.configfile.settings.printer.kinematics is in ['delta','polar','rotary_delta','winch']
                else False %}
  ##### define park position #####
  {% set z_min = params.Z_MIN|default(0)|float %}
  {% set z_park = [[(act.z + park_dz), z_min]|max, (max.z - origin.z)]|min %}
  {% set x_park = params.X       if params.X is defined
             else custom_park_x  if use_custom
             else 0.0            if round_bed
             else (max.x - 5.0) %}
  {% set y_park = params.Y       if params.Y is defined
             else custom_park_y  if use_custom
             else (max.y - 5.0)  if round_bed and z_park < cone
             else 0.0            if round_bed
             else (max.y - 5.0) %}
  ##### end of definitions #####
  _CLIENT_RETRACT
  {% if "xyz" in printer.toolhead.homed_axes %}
    G90
    G1 Z{z_park} F{sp_hop}
    G1 X{x_park} Y{y_park} F{sp_move}
    {% if not printer.gcode_move.absolute_coordinates %} G91 {% endif %}
  {% else %}
    RESPOND TYPE=echo MSG='Printer not homed'
  {% endif %}

PhilBaz, 2023-10-05:
@meteyou Sorry, nevermind.....just saw there was a credit at the top.