mainsail-crew/mainsail #1597: Spoolman French translation added

Bahoue, 2023-10-08:
### Requested feature:

  spoolman translation added to fr.json      

        },
        "SpoolmanPanel": {
            "Cancel": "Annuler",
            "ChangeSpool": "Changer de bobine",
            "DaysAgo": "{days} jours avant",
            "EjectSpool": "Ejecter la bobine",
            "EjectSpoolQuestion": "Êtes-vous sûr de vouloir éjecter la bobine de filament?",
            "Filament": "Filament",
            "FilamentTypeMismatch": "Le matériau de la bobine active ({spoolType}) ne correspond pas au matériau du G-Code ({fileType}).",
            "FirstUsedOutput": "Première utilisation: {firstUsed}",
            "Headline": "Spoolman",
            "LastUsed": "Dernière utilisation",
            "Location": "Localisation",
            "Material": "Matériau",
            "Never": "Jamais",
            "NoActiveSpool": "Le suivi du filament est inactif. Pour commencer, veuillez sélectionner une bobine.",
            "NoResults": "Aucune bobine trouvée avec les critères de recherche actuels.",
            "NoSpools": "Pas de bobines disponibles",
            "NoSpoolSelected": "Aucune bobine sélectionnée. Veuillez sélectionner une bobine ou cette impression ne sera pas suivie.",
            "OpenSpoolManager": "ouvrir le gestionnaire de bobines",
            "Refresh": "rafraîchir",
            "Search": "Recherche",
            "SelectSpool": "Sélectionner la bobine",
            "Today": "Aujourd'hui",
            "TooLessFilament": "Il se peut que la bobine actuelle ne contienne pas assez de filament pour cette impression. ({spoolWeight}g pour {fileWeight}g)",
            "Weight": "Poids",
            "Yesterday": "Hier"
        },

[fr json.zip](https://github.com/mainsail-crew/mainsail/files/12841253/fr.json.zip)

meteyou, 2023-10-09:
Thank you! Can you make a PR?

Bahoue, 2023-10-09:
> Thank you! Can you make a PR?

I would have done it with pleasure, but I don't know how to do it.

meteyou, 2023-10-12:
merged with #1597

closed 2023-10-12 by meteyou