mainsail-crew/mainsail #1600: When moonraker is built with `--base`, it still tries to get config.json from /

Laikulo, 2023-10-10:
### Mainsail Version:

g932a2c35

### Browser:

Firefox

### Device:

Laptop

### Operating System:

Windows

### What happened?

After building mainsail with --base=/mainsail/, and serving that at /mainsail/ on an httpd: When browsing to /mainsail/, mainsail tries to retrieve config.json from /

### What did you expect to happen instead?

Mainsail to retrieve config.json from /mainsail/

### How to reproduce this bug?

1. `npm ci`
2. `npm exec vite build --base=/mainsail/`
3. `cp base/* -r /srv/www/mainsail/`
4. `firefox localhost/mainsail/`

meteyou, 2023-10-10:
Mainsail don't support the base attribute. But there is a PR which will add something like this.

Laikulo, 2024-01-15:
Here's an example of a workaround with Caddy, that allows mainsail to work as normal:
```
@mainsail_config_bodge {
  path /config.json
  header Referer */mainsail/*
}
redir @mainsail_config_bodge /mainsail/config.json permanent
```

PR #1873 (2024-06-09): feat: add support for base url
## Description

This PR allows to use a base url, when building Mainsail. This PR cannot fix a dynamic subdirectory URL. So you have to build your own Mainsail build for each subdirectory.

Maybe it is possible to release a second build in the release for `/mainsail` as subdirectory, but we have to double-check this, to have no issues with the Moonraker update manager.

## Related Tickets & Documents

fixes #1600 
fixes a part of #1163 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2024-06-09 by meteyou