mainsail-crew/mainsail #1601: feat: Light control

stijnb1234, 2023-10-10:
### Requested feature:

Add an official way to control one or more lights of the printer, even when the printer itself is not powered on.

Currently, it's possible to add a button under 'Others', but it's not that easy to use and is also not available when the printer itself is offline.

I would like my light to stay on while the printer is turned off so that I can see the finished print and turn off the light myself afterward.

meteyou, 2023-10-11:
how should the light work when the printer is off?

Mainsail itself is only the web interface and it only communicate via API. so when your printer is OFF, there is no possibility to "control" anything.

closed 2023-10-11 by meteyou

stijnb1234, 2023-10-12:
> how should the light work when the printer is off?
> 
> Mainsail itself is only the web interface and it only communicate via API. so when your printer is OFF, there is no possibility to "control" anything.

As long as the host is online, it's possible to control, for example, lights (with GPIO) connected to the host?

meteyou, 2023-10-12:
Then you have to use the Moonraker power devices