mainsail-crew/mainsail #1603: Issue when installing on Raspberry Pi OS Bookworm (64Bit) (Used KIAUH)

phuzzyday, 2023-10-12:
### Mainsail Version:

2.8.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

The above choices re:Browser, Device, OS, are N/A, as ythe problem occurred during installation. Noticing now that it seemed to be compiling camera-streamer.

Please see end of the log here: https://pastebin.com/gaGAa7uw

### What did you expect to happen instead?

Error free installation

### How to reproduce this bug?

RPI4 8Gb. Installed fresh OK as noted.
Immediately loaded KIAUH instruction page, installed Klipper, Moonraker, Mainsail, Crowsnest. Error occured during Crowsnest install.

meteyou, 2023-10-12:
sry... there are multiple issues with this issue.

Mainsail itself is just the webgui, crowsnest is a different repo + has no bookworm support + KIAUH has nothing todo with both of it + camera-streamer is an external lib, which will only installed via crowsnest + raspi pi os bookworm has no libcamera support, so it cannot work (right now).

closed 2023-10-12 by meteyou

phuzzyday, 2023-10-13:
Thanks anyhow! As long as someone knows! Where would you recommend I post a note? Camera streamer?

meteyou, 2023-10-15:
@phuzzyday there is already an open ticket: https://github.com/ayufan/camera-streamer/issues/107

meteyou, 2023-10-15:
ohhh... just saw, that you open the ticket yourself.

ayufan, 2023-10-17:
The bookworm is supported by `camera-streamer` if installed via prebuilt: https://github.com/ayufan/camera-streamer/releases/tag/v0.2.7.