mainsail-crew/mainsail #1616: [feature request] Graph for speed and flow

tippfehlr, 2023-10-20:
### Requested feature:

optionally display the speed and flow data in a chart.

Quick mockup I did:
[image]

just a little expandable section

### Solves the following problem:

Could help with seeing to which speeds the printer really accelerates with high speed and locked acceleration.

meteyou, 2023-10-20:
Moonraker has no store for these values. So every time, you refresh the tab or close/open mainsail, the chart would start from 0.

FoldZero, 2024-05-12:
Fliud now has 'Diagnostics' which is basicly this.

meteyou, 2024-05-12:
@FoldZero yes, but this start every times you open webinterface again. as long, as there is no cache in moonraker to store it, we will not implement it.