mainsail-crew/mainsail #1624: BED_SCREWS_ADJUST range error for position max

oernster, 2023-10-23:
### Mainsail Version:

2.8.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

If you set your stepper_x position_max or stepper_y position_max to values greater than those in the bed_screws section in your printer.cfg then when you send the command BED_SCREWS_ADJUST, the UI popup appears but you cannot do anything.  The only way to recover is to reboot the printer and adjust the values suitably in the bed_screws section in your printer.cfg.
Perhaps there should be a warning to fix your bed screws to be within ranges?  I have reported this on the klipper discord but I strongly suspect there will need to be a supporting adjustment to the UI in mainsail also.

Thanks for your great work guys; love the software.

### What did you expect to happen instead?

Allow ANY button on the UI popup to do something.

### How to reproduce this bug?

See what happened.

godofal, 2023-10-30:
Not a fix but a faster reset: SSH into your pi and send the following to restart klipper instead of the whole printer/pi.

> sudo systemctl restart klipper

pedrolamas, 2023-11-01:
I've pushed a Pull Request to Klipper to fix this issue as I have confirmed it is a bug (tool gets stuck in execution mode).

meteyou, 2023-11-01:
@pedrolamas ohh. thank you very much! i haven't had time to take a closer look in Klipper.

pedrolamas, 2023-11-02:
My PR has been merged, so I think this issue is now fixed upstream in Klipper!

meteyou, 2023-11-02:
@pedrolamas Thx! I can confirm that it works as expected!

[image]

closed 2023-11-02 by meteyou