mainsail-crew/mainsail #1625: Add support for action command prompts

ulope, 2023-10-24:
### Requested feature:

Octoprint has had support for [Action Command Prompts](https://docs.octoprint.org/en/master/bundledplugins/action_command_prompt.html) for quite some time now. 

It would be great if Mainsail also supported it.

### Solves the following problem:

They are very useful for user interactive macros, e.g. filament loading / unloading, color change, confirmation prompts, etc.

### Additional information:

The protocol is quite simple and described on the linked octoprint page.

meteyou, 2023-10-24:
Octoprint is quite different to mainsail. Octoprint is server and webclient at the same time. Mainsail is only a webclient. So this is not possible without any change on the server side. Mainsail only supports Klipper macros to start this command.

ulope, 2023-10-24:
Hm yeah alright that makes sense from an implementation standpoint, nonetheless I think this would be a great usability win in general.

Would you be open to implementing this if server support (i.e. moonraker) were available?

meteyou, 2023-10-24:
Ok... We spent the half evening discussing this feature internally with the crew, and I was probably a bit hasty in closing it. Maybe we have found a way to implement this. I will reopen this FR and test it before I can give feedback.

ulope, 2023-10-25:
Awesome!
I also looked a bit more into this yesterday and it looks to me that the `RESPOND` messages are already coming through to the frontend via the `notify_gcode_response` websocket messages. 

So it should "just" be a matter of parsing the `// action: prompt_...` messages.
(And yes I know that "just" is a dangerous word in software development ;))

meteyou, 2023-10-25:
yeah, thats my idea. Klipper use this "commands" already for talking with octoprint. so i can use it the same way and "just" parse the console output for these commands.