mainsail-crew/mainsail #1626: [FR] Ability to expose native Klipper commands as buttons

Sineos, 2023-10-24:
### Requested feature:

Just stumbled over this example: https://klipper.discourse.group/t/internal-error-bed-screws-adjust/11159/2

I find it horrid to create a macro just to be able to display built-in Klipper commands as a button in the UI.

### Solves the following problem:

Provide means to map built-in Klipper commands to buttons to avoid messing with the printer.cfg in such ways.

meteyou, 2023-10-24:
This is already implemented since v2.2.0 and there is also a "helper dialog" with the results after you execute this "macro"/function.

Sineos, 2023-10-25:
Please excuse my ignorance, but I'm not able to find it.
As far as I can tell, to create a button in the UI, I need to define a macro in `printer.cfg`. Going by the example linked above, the user defined a macro, e.g.:

```
[gcode_macro Accept_Bed]
gcode:
    ACCEPT
```

So, a macro needs to be defined only to map a built-in Klipper function to a button, which I find will only create clutter in the printer.cfg and "abuses" it to modify the UI.

What I wanted to propose is some means to map built-in functions (i.e. as defined in https://www.klipper3d.org/G-Codes.html) to a button without the need to put it into a macro.

meteyou, 2023-10-25:
Np. So the "start button" is in the toolhead panel, in the 3-dots submenu (all leveling functions should be here, when they are configured). After the start, mainsail should open a dialog with the result and a accept and abort button.

Sineos, 2023-10-25:
Ah, many thanks. I was aware of this "context related" options.

My point was more general in nature, e.g. allowing to create a button that calls `BLTOUCH_DEBUG COMMAND=pin_down` or `GET_POSITION` without having to resort to defining it in a macro

meteyou, 2023-10-28:
ah ok. yeah we try to add all of these commands, but for some (like screw tilt adjust, z_calibraten, ...) we also need some functions in klipper itself, that we can get the right response from klipper to display it the right way.

`BLTOUCH_DEBUG` is currently missing in mainsail. i don't have/use a BLTOUCH, so its difficult for me to implement something for it. if you can help me with the concept and do some tests, i would be very happy to implement something like that.

`GET_POSITION` should be "implemented" in the toolhead panel. or all important values for a "normal user". what values would you personally still be missing?

if you know of any other such commands, i would be happy to talk about all of them to see what else mainsail is missing. if we can/should add a support for these commands, i would ask you to create a separate FR for each single task/command. so i have everything on my "todo list".

Sineos, 2023-10-28:
Personally, I'm missing no commands, thanks 👍 
I just stumble over such "one command macros" in the support activities quite often and I believe that a macro is the wrong way of defining an UI button.

I think that the old Octoprint veteran has solved this quite nicely with its [Custom Controls](https://docs.octoprint.org/en/master/features/custom_controls.html) for which also a plug-in exists that allow defining them quite easily.

meteyou, 2023-10-28:
this is exactly what we wanted to avoid, as it is probably difficult for users to understand why macro code (jinja) does not work now and why there are 2 ways to create "macros" at all. furthermore, quite a lot of "one line macros" are simply copied together or "legacy" code from macro example files.

i don't think we need to implement every single command/gcode of klipper in the gui, but every command/gcode which users need to use for daily use (like all the bed level commands).

but i would still appreciate if you keep me informed if you see something you can't assign in the GUI, or if you would like to see it implemented as a button.

Sineos, 2023-10-28:
Fair point of view.
If I should notice something apparent, I'll report back.

zellneralex, 2023-10-28:
My 2 cents here, we have a console with autocomplion, so that these rare used commands can be entered very easy there.
As a homework for Klipper: get a Methode in to show the valid input parameter …
Especially the bltouch debug command shows the problem here. The user needs to open the Klipper reference to see why parameter he can use.

dw-0, 2023-10-28:
Don't mind the old version of Mainsail in the following clip, but just out of curiousity, would this be kind of the implementation you are looking for?

https://github.com/mainsail-crew/mainsail/assets/31533186/36acfc8a-9c08-48b8-a724-182936f2d697

Sineos, 2023-10-28:
@dw-0 this basically is exactly the solution I had in mind.

Maybe as a separate "Button" Panel or together with the Macro buttons but in a different color to distinguish them.

zellneralex, 2023-10-28:
Honestly, where is the difference? Adding 2 lines in the Klipper config or to adding 2 lines in the UI?

I see it that we making it more complex for the user as he then has a second place to search for. At the octroprint times the mentioned plugin was the only way to get anything as a button in the UI so worst case you needed to do the macro and than add it to custom control. 

Currently we have a strict rule, all macro buttons are macros defined in the printer config

Sineos, 2023-10-28:
As already answered to @meteyou, this is perfectly fine for me, since I do not need either personally. 
As you have mentioned @zellneralex there is the console autocompletion, which I enjoy very much and you are also right that some of the commands always need referring back to the documentation for the proper syntax / parameters, which is cumbersome.

Just seeing all these "button macros" in the support activities made me think if defining a Klipper macro to create a button in the UI is the right approach. IMO it is not and rather a crude workaround. Apparently the users like to do it and this is why I have placed this FR.

greatwaldini, 2023-11-07:
> i don't think we need to implement every single command/gcode of klipper in the gui, but every command/gcode which users need to use for daily use (like all the bed level commands).
>
>
>
> but i would [...]

I thoroughly agree with these  design principles.  Yet still, I can empathize with the feature request.

I ended up on this issue page tonight while searching for a way to solve the root of OP's frustration, namely a small but persistent friction present in Mainsail. (First thing I read when doing so, so perhaps the following already exists..)

What I feel a sore need for is a way to open .cfg files from the dashboard.  I think this would satisfy what OP is looking for too, indirectly.

I keep my macros in a dedicated .cfg file, which I then include in my printer.cfg.  Whenever I need to edit or add a macro or change some parameter in my config, I encounter a bit of friction as I navigate to the 'Machine' tab and then scroll through a bunch of files looking for the one I want.  It's often one of two.

Now I'm fairly new to 3D Printing, so I don't know about everyone else but I'm in and out of printer.cfg and the macro cfg _all the time._

So if I could have the ability to navigate directly to the .cfg in the code editor for files I interact with directly and frequently, that would be a beautiful thing.

These direct links into the editor for a specified file could be implemented as either dashboard buttons or could populate in the navigation menu itself?  I think I would personally prefer buttons on the dashboard, because with this addition I would be navigating to the Machine tab about as often as I navigate to the other tabs (Webcam full screen, Gcode viewer, etc) - rare to never.

drewgwallace, 2023-11-19:
I would like to be able to add existing commands as a "button", that is of the same name as the command. Something like this macro will not work in a printer.cfg, it must be a different name than something already registered:

```
[gcode_macro PROBE_CALIBRATE]
gcode:
  PROBE_CALIBRATE
```

My reasoning for wanting it to be the same name is now I have a chance to reinforce memory of the correct gcode.

meteyou, 2023-11-19:
@drewgwallace i don't think i will implement something like that. so you cannot create a macro with the same name, i don't want to create "virtual macros" in mainsail.

I already explained above, why i don't want todo that.

meteyou, 2024-01-04:
i will close this post, because the main question was answered and all other questions here are not related to the original request.

closed 2024-01-04 by meteyou