mainsail-crew/mainsail #1632: [MacOS]: Websocket infinite reconnection for locally hosted instance, all browsers are affected

SpaghettiC0des, 2023-10-30:
### Mainsail Version:

2.8.0

### Browser:

Safari

### Device:

Laptop

### Operating System:

macOS

### What happened?

The issue was initially posted here https://github.com/Arksine/moonraker/issues/726 and https://github.com/Rat-OS/RatOS/issues/106.

Similar issue from a guy on Reddit, https://www.reddit.com/r/klippers/comments/16u2jt9/cant_connect_to_moonraker_ip_on_mac_but_can_on/?utm_source=share&utm_medium=web2x&context=3

### What did you expect to happen instead?

It should work on MacOS Sonoma

### How to reproduce this bug?

Open the `.local` DNS for example mine is `ratos.local` in any browser using MacOS Sonoma.

### Additional information:

- it works on an iPhone
- it works when using http://my.mainsail.xyz

meteyou, 2023-10-30:
Which port do you use to connect to your printer on my.mainsail.xyz?

SpaghettiC0des, 2023-10-30:
It’s the default 7125

meteyou, 2023-10-30:
What happen, if you use port 80?

SpaghettiC0des, 2023-10-30:
When I connect to port 80, it will not be able to connect to the websocket.

meteyou, 2023-10-30:
Pls upload you moonraker.log and your nginx config.

meteyou, 2023-10-30:
i double-checked your conversation with arksine in the moonraker repo. i also tested today all my mac devices and i cannot find any issue.

my mac mini is running 13.5.1 -> without issues
macbook air was running 14.0 -> without issues
macbook air updated to 14.1 -> without issues

i tested on every os version safari + chrome

Arksine, 2023-10-30:
Meteyou informed me in the Moonraker issue that Fluidd connects to port 7125 by default.  Given this, it sounds like its an issue with nginx.

In addition to the log files requested by meteyou, I would recommend double checking to see if `mainsail-access.log` and `mainsail-error.log` are not available.  The error log in particular may provide relevant information.

[image]

If they aren't there then we should be able to get them via scp.  I'm not familiar with MacOS but I presume it is very similar to linux.

In a future version of MainsailOS it might be worthwhile to provide a symbolic link to the nginx configuration in the `~/printer_data/config` .  We wouldn't be able to modify it, but we should be able to read it and therefore download it.

meteyou, 2023-10-31:
@Arksine we have now another user on our discord server with the same issue. He only create a new user on the identical Mac and it works with the new user. So it's an issue with one specific MacOS version with some user settings, but we don't know which...

SpaghettiC0des, 2023-10-31:
@meteyou I'm using the latest MacOS Sonoma. My M1 Mac is not working; my 2016 Intel MacBook was able to access the printer's local DNS but is using an old MacOS version.

meteyou, 2023-10-31:
@SpaghettiC0des i tested it yesterday with my MacBook air M1 with the newest updates and I can't reproduce this issue.

SpaghettiC0des, 2023-10-31:
@meteyou, thanks for checking! I have only one app installed that modifies the network settings, freedom.to. But I don't think it's causing the issue because I turn it off when using the printer. 

Local DNS' port `80` seems to be the only port affected by this issue.

SpaghettiC0des, 2023-10-31:
[mainsail-access.log](https://github.com/mainsail-crew/mainsail/files/13217983/mainsail-access.log)

@Arksine @meteyou, here are the `mainsail-access` logs, the error log is empty, so I will not include it here.

meteyou, 2023-12-08:
I cannot reproduce this issue and there are also no other users with this bug. So it look like an issue with your specific setup. I will reopen this post, if there are more users with this bug and it is possible to reproduce it.

closed 2023-12-08 by meteyou

SpaghettiC0des, 2024-03-30:
@3DCoded @meteyou 

I found the issue and found a solution.

The issue is that the Raspberry Pi's OS(Ubuntu) datetime, which is out of sync and it is causing some issues with the SSL(?).

Try running `sudo ntpdate pool.ntp.org`; if you encounter a `ntpdate command not found error,` you must install it first.

meteyou, 2024-03-31:
the system time, should only be important with SSL/https. But you have to keep in mind, when you connect from https://my.mainsail.xyz you have also to use moonraker with SSL, because your browser only allow to connect from a SSL website to a SSL secure websocket.

3DCoded, 2024-04-01:
> @3DCoded @meteyou
>
> I found the issue and found a solution.
>
> The issue is that the Raspberry Pi's OS(Ubuntu) datetime, which is out of sync and it is causing some issues with the SSL(?).
>
> Try running [...]

@SpaghettiC0des 
I tried your solution and I am still unable to get Mainsail to work with my Mac. When I run the command, I get the following output:
`2024-04-01 15:52:45.229046 (-0400) +0.497748 +/- 0.510828 pool.ntp.org 23.157.160.168 s2 no-leap`
However, after running it a couple more times, I get this output:
`2024-04-01 15:52:57.241842 (-0400) +1.014783 +/- 1.031116 pool.ntp.org 23.157.160.168 s2 no-leap
CLOCK: time stepped by 1.014783`
and then:
`2024-04-01 15:52:59.998458 (-0400) -1.022499 +/- 0.008533 pool.ntp.org 23.157.160.168 s2 no-leap
CLOCK: time stepped by -1.022499`

Any help would be appreciated.

meteyou, 2024-04-02:
@3DCoded do you use SSL to connect to mainsail?

3DCoded, 2024-04-03:
I don't think I use SSL to connect to Mainsail. I just used the default install from KIAUH.

meteyou, 2024-04-04:
I think the system time is only important for SSL, so this should not be a problem in your case.

jpwolfe9912, 2024-08-11:
@SpaghettiC0des Did you ever figure this out? I'm having the same issue on Mac

3DCoded, 2024-08-11:
I ended up wiping my Pi several times and reinstalling my VPN to make it work.

meteyou, 2024-08-11:
@3DCoded to be honest. It is no issue with the Pi. its just a MacOS/Safari issue. so wiping the pi will not fix this issue...

jpwolfe9912, 2024-08-11:
I've had other weird network issues on my Mac where it just won't connect to certain websites or servers that my phone or PC have no problem with. Is there some way to hard reset a Mac's network settings? I couldn't find anything good online

meteyou, 2024-08-11:
@jpwolfe9912 if this would be a generic "mac network issue", then we would also have issues with other browsers, but this issue only exists with Safari (and maybe only in one single version of it)

3DCoded, 2024-08-11:
> @3DCoded to be honest. It is no issue with the Pi. its just a MacOS/Safari issue. so wiping the pi will not fix this issue...

I'm not exactly sure why wiping the Pi fixed it. My installation of Mainsail may have been corrupted. For some reason, it still only loads now if I install Fluidd on port 80, and Mainsail on Port 81. This also happens with Chrome.

meteyou, 2024-08-11:
@3DCoded then you have a different issue. This issue here is only a Safari issue.

jpwolfe9912, 2024-08-12:
Part of the issue might be with the Pi but I think it's mostly the Mac. Doesn't work on any browser. Weirdly enough, if I use my Windows VM on my Mac, it works fine... isn't that weird.

meteyou, 2024-08-12:
@jpwolfe9912 then it is a different issue then this issue here. Feel free to open a post in the support-forum on our discord or here in the discussions section.

This issue here is only for the issue between safari on Mac and your host.

meteyou, 2024-08-17:
we found a possible reason for this issue. pls check the post in the "new" issue and pls give some feedback: https://github.com/mainsail-crew/mainsail/issues/1774#issuecomment-2294825913