mainsail-crew/mainsail #1633: Mainsail won't load if not able to connect to spoolman

smwoodward, 2023-10-30:
### Mainsail Version:

2.8

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

If the hosted computer that has spoolman running on it is offline, mainsail won't load

### What did you expect to happen instead?

For mainsail to load even if spoolman is not reachable

### How to reproduce this bug?

Configure moonraker/mainsail to connect to spoolman, and turn off spoolman and try to reload mainsail.  This was also while the printer was running.

hawkeyexp, 2023-11-03:
Confirmed - had the same behaviour some days ago while spoolman was not started after a nas reboot

smwoodward, 2023-11-03:
Good that it's not just me.

meteyou, 2023-11-03:
Thx for the report. This is a known bug and i have to add some "workflow" for init plugins, which can fail (like this one).

mlee12382, 2023-11-04:
I am also experiencing this bug. Glad that's all that's causing the issue. I was able to load Fluidd but Mainsail will get stuck on initializing when the Spoolman host is offline. I hadn't figured out the cause until I saw this issue and commented out the Spoolman section in moonraker.conf and Mainsail loads properly so issue cause confirmed.

PR #1680 (2023-12-10): feat: add moonraker init component check with warning
## Description

This PR adds a check for failed Moonraker components, skip  the Initialization dialog and adds a warning for this component. A typical issue, when you need this feature is, when Moonraker has no connection to spoolman. Then Mainsail has to skip the init process of this module.

## Related Tickets & Documents

fixes: #1633 

## Mobile & Desktop Screenshots/Recordings

[image]

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2023-12-10 by meteyou