mainsail-crew/mainsail #1643: File browser: add ability to quickly jump to any path segment

mdziekon, 2023-11-06:
### Requested feature:

Current implementation of the files browser (aka "G-Code files") does not have a quick way of going back a couple of directories back, eg. in situation where I'm at "/some/models/are/located/here", and I would like to quickly go back to "/some/models". Instead, users have to click a couple of times on the ".." (go back) directory icon. This can get pretty tedious when dealing with elaborate gcode file trees.

To better visualise what's this proposal is about, take a look at the following "before" and "after" clips:

BEFORE:
[before_quicknav.webm](https://github.com/mainsail-crew/mainsail/assets/4425221/37a4ddb5-0b14-472a-ab43-bcda450a03a7)

AFTER:
[after_quicknav.webm](https://github.com/mainsail-crew/mainsail/assets/4425221/7337d4e6-2ebd-437c-a2b5-a8aa9db654bb)

Adding the ability to quickly jump back would be a nice UX improvement, and would bring some feature parity to usual file browsers found eg. on users' usual desktop environments.

### Additional information:

I'm willing to submit a PR with the feature already implemented, provided the FR gets accepted by the maintainers.

meteyou, 2023-11-14:
sorry for the late reply! i would really like this feature, the only thing i would implement differently would be to use the word "/gcodes" instead of the house icon, as this is the real path.

If you make a PR, could you add this to the config files at the same time?

closed 2023-12-01 by mdziekon

meteyou, 2023-12-02:
Yes and thanks again! Normally, GitHub closes linked issues automatically.