mainsail-crew/mainsail #1648: System Load (CPU Usage) Panel on Dashboard

Jmoore1127, 2023-11-11:
### Requested feature:

I have been looking for a way to add the system load guages (from the Machine page) to the the dashboard. I am often trying to use advanced features with a somewhat limited setup and being able to monitor usage while a print is running would help to judge how much of an impact recent changes have had. I haven't  been able to find a way to add these yet which I found surprising.

Is it possible to add the CPU and MCU usage guages to a panel on the dashboard? Even better if they were a line chart similar to the temperature chart.****

### Solves the following problem:

Constantly switching between dashboard and the machine pages

ModernMakerStudios, 2024-04-28:
i am looking for information on this as well.

meteyou, 2024-05-17:
@sl0m0ZA what do you mean?