mainsail-crew/mainsail #1650: Browser back misbehaves when code editor is open.

fjpanag, 2023-11-12:
### Mainsail Version:

v2.8.0

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Linux

### What happened?

I navigate to the machine tab and I open a `.conf` file with the text editor.

When I click the back button of the browser, the editor remains open but Mainsail navigates to the previous tab in the background.

From the user perspective, nothing happens, but when you eventually close the editor you find yourself in an unrelated tab, other than the machine one.

This seems as a non-issue (just don't click back...), but in fact my mouse has 5 buttons, including back/forward, and out of habit I constantly click back and mess things up...

### What did you expect to happen instead?

When the editor is open, clicking back should close the editor.  
Either with discarding the changes, or by asking whether they should be saved.

Then, after the editor is closed, the next press of the back button should behave as usual.

*(Alternatively, the back button may do nothing when the editor is open, but I would prefer the above solution.)*

### How to reproduce this bug?

1. Go to machine tab.
2. Open a file with the editor.
3. Press the back button of your browser or mouse.

The editor is still open.  
If you close it, you are no longer in the machine tab.

meteyou, 2023-11-12:
This is quite normal, because the editor is no "new site" in the history. its only an overlay.

closed 2023-11-12 by meteyou

fjpanag, 2023-11-12:
> This is quite normal, because the editor is no "new site" in the history. its only an overlay.

I understand that, but usually you have the option to click outside an overlay to close it.

Since the editor is full-screen, from the user perspective it is just a new page.  
It is an overlay only technically.

So although the current implementation is "technically" correct, as UX it is suboptimal...  
Maybe at least disable back when an overlay is displayed?

fleaz, 2025-01-04:
Hey, 
sorry for digging up this old issue, but I also run into this "issue" all the time and think that we could do better here to make the usage of mainsail more pleasant.

 I see the point from meteyou that it's **technically** correct what mainsail is doing here, but just like fjpanag wrote, that's a UX nightmare for the user. Because the editor overlay is nearly fullscreen, this *feels* like a new page even though it isn't and my brain automatically tries to close the editor with the back button on my mouse, then I remember that I have to click the X and then I'm already back on the dashboard and no longer in the machine tab because the rest of the mainsail website already handled my many back button presses :D

Mapping the "back" event from the browser to "close editor" in this scenario, would be a really great improvement imho

meteyou, 2025-01-04:
i don't know any way to detect this "back event". i think it's just a "browser history" function. i haven't read up on this either, but without doing any detailed research i would say that it's not possible (current state of knowledge)