mainsail-crew/mainsail #1652: Rename files wont record print statistics in G-Code Files

icariantk, 2023-11-13:
### Mainsail Version:

2.8.0

### Browser:

Chrome

### Device:

Other (Please tell us below)

### Operating System:

Linux

### What happened?

If i rename the uploaded file via UltiMaker Cura Moonraker plugin, the G-Code file manager will not show the statistics of the print.

This is the file in the history tab
[image]

But in G-Code menu shows:

[image]

The rename was before priting. I tested with the default name that Cura Uploaded the file and the "Queen" got the statistic record of the G-Code tab.

### What did you expect to happen instead?

The statistic of the file keeps showing/recording even if the rename of the file in the G-Code tab.

### How to reproduce this bug?

My instalation is in an ARMBIAN TVBox, so it might have something to do with it, but i doubt it very much since the UI is working as expected if no rename is made.

### Additional information:

Using CURA 5.5 and Moonraker Connection 1.7.2
Mainsail was installed using KIAUH Script.

meteyou, 2023-11-13:
Statistics and G-code metadata are more a part of Moonraker and not mainsail. Mainsail should only display it, if they exist.

Pls tell me exactly which Cura plugin you use. There are more than one available, that I can double check it.

icariantk, 2023-11-13:
I see, i will double check the moonraker configuration in the meantime, this is the plugin im using:

https://marketplace.ultimaker.com/app/cura/plugins/emtrax/MoonrakerConnection

meteyou, 2023-12-23:
i cannot reproduce this issue on my printers. is it possible, that your uploaded the file again? normally each file will get a ID , when you upload a new one. this ID will be checked with the history for this number.

meteyou, 2023-12-31:
i will close this PR, because of no response. If there are other users with the same issue, pls post a guide how to reproduce this issue and i will reopen this issue again.

closed 2023-12-31 by meteyou