mainsail-crew/mainsail #1653: Bed temperature not restored after Pause

trophysoft, 2023-11-13:
### Mainsail Version:

2.8.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

If the print is paused and then resumed, Mainsail restores the nozzle temperature but not the bed temperature.

### What did you expect to happen instead?

Restore the bed temperature as well.

### How to reproduce this bug?

Pause the print, wait for a while to drop the bed temperature and then resume the print.

zellneralex, 2023-11-19:
If your bed cools down your print will loose bed adhesion. So you need to insure that the bed stays on. See the idle_timeout example < https://github.com/mainsail-crew/mainsail-config#new-feature-saverestore-extruder-temperature-on-pauseresume>

And you are at the wrong GitHub, as that stuff is handled in <https://github.com/mainsail-crew/mainsail-config>

meteyou, 2023-11-19:
i will transfer this issue to [mainsail-crew/mainsail-config](https://github.com/mainsail-crew/mainsail-config)...