mainsail-crew/mainsail #1654: Case light toggle in webcam widget headers

hellbe, 2023-11-14:
### Requested feature:

It would be fantastic to have a on/off toggle for printer leds in the webcam widget header. I don't want to leave lights on constantly, checking in on the print via webcam from the phone it would be super conventient to toggle the light on and off right there. Would suggest design similar to the print pause button.

Something like this, but cleaner design ofc :)
https://www.dropbox.com/scl/fi/ncwcw09ehqfvrnac9s5r9/mainsail-light-toggle.png?rlkey=ulq13dlcd5133o4fmoivke9dx&dl=0

meteyou, 2023-11-14:
thank you for your feature request! something like this has already been requested several times (but mostly with the option in the title bar).

how would you assign the right light to this button?

hellbe, 2023-11-15:
@meteyou would suggest adding a field to the webcam settings referencing the output pin name/id in printer.cfg. Example:

[image]

hellbe, 2023-11-15:
Alternative placement could be overlay on the webcam image. Would enable individual control when showing multiple webcams. But, I think the title bar is nicer and in multi-view it could toggle all configured lights for the shown cams.

[image]

meteyou, 2023-11-15:
Output pin is the old variant... Today, led or neopixel is more recommended for caselight.

hellbe, 2023-11-15:
Ah right, screenshot is from default Voron Trident config I am running. Will update.

Given the better modern config entities, maybe a good solution is a drop-down with all led config entries?

https://www.klipper3d.org/Config_Reference.html#leds

jmerdich, 2024-02-17:
If you're going through with this, I'd much prefer allowing any macro here instead of directly taking klipper's leds. A lot of lights go through Moonraker's integrations and not through klipper itself for various reasons (WLED niceties, too many leds, etc.). Even within klipper, you may want multiple lights to turn on, only turn it on to XX% power, or apply an led template, and that sounds very complicated to handle within mainsail but very easy to leave up to a macro.

For those that have the simple case and use klipper's built in support for leds, it's not hard to write a macro that just does SET_LED, and having a simple 'LIGHTS_ON/OFF' macro benefits other places like klipperscreen. Alternatively, you can let the macro in mainsail take parameters and default to something like 'SET_LED LED=lights WHITE={brightness}'

hellbe, 2024-02-17:
Sounds like a great suggestion @jmerdich