mainsail-crew/mainsail #1658: Add filament weight to history / statistics

EmJay276, 2023-11-18:
### Requested feature:

The filament weight is already displayed at the Dashboard and the G-Code Files, it would be realy nice to also have this value for the statistics in stead of the filament length.
[image]
[image]

[image]

meteyou, 2023-11-18:
this is not possible, because it only exists in the metadata from the filament, but not in the history data of moonraker. this data comes from klipper, and not from the metadata. Klipper dont know the specific filament weight. it can only track the extruder (used filament length). you would need some hardware to weight the filament, that klipper would know the "used filament weight". so this is not possible.

Here are the current available data:
[image]

EmJay276, 2023-11-18:
Using the metadata, it would be easily possible to calculate a "length-density" from the Gcode.
`density_length = metdata.filament_weight_total / metadata.filament_total`
This density can be used to calculate the total weight of the print
`weight_used = density_length * filament_used`
In your example the length density would be `~2.934464205524863719639451695975 g / m` and the total filament used `~29.646g` for the `10.10277m`

Of course, there is an error if faulty settings are made in the slicer, but that's also the case for the length, if the extruder steps are not calibrated.
I would find it really handy to have a rough estimate of the filament (weight) used in a print.

meteyou, 2023-11-18:
thank you very much! i hadn't thought of that. of course it would be possible. i'll try to implement it.

EmJay276, 2023-11-18:
That would be great! If you need any help, let me know.

meteyou, 2023-11-23:
@Tom-Neverwinter ohh... I found it. Your slicer is just wrong. This feature request is for history and not gcode files!

m-klecka, 2024-02-12:
Any updates on this? Filament used in grams in the print history would be useful, instead of meters which tell basically nothing..

meteyou, 2024-02-12:
I'm refactoring the complete history page/list. after that i can/will implement this feature.

TMTYD, 2024-04-19:
Why is the filament weight not displayed?

[image]

[image]

meteyou, 2024-04-19:
@TMTYD your question has nothing Todo with this FR. Plus use our discord or discussions forum for help. (I think it will be a slicer setting issue, but your screenshots are useless for debugging)

LancerGM, 2024-06-26:
This would be an extremely good feature to have! Currently I am running a filament manager which I will not need anymore once this feature is implemented.