mainsail-crew/mainsail #1665: Documentation: adapt to change in label objects option in PrusaSlicer

tweh, 2023-11-25:
### Mainsail Version:

2.8.0

### Browser:

Safari

### Device:

Laptop

### Operating System:

macOS

### What happened?

I’m not sure whether this is the right place to report this since this is a thing that should be change in the documentation and not a real bug. With version [2.7.0 Prusa Slicer](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.0-alpha1) changed it’s option `Label Objects` to become a drop down and thus the screenshot in the [mainsail docs ](https://docs.mainsail.xyz/overview/features/exclude-objects). So at the moment the docs show only a check box, where as far as I understand one must 1. select *Firmware-specific* and 2. in the Print settings and *Klipper* as g-Code flavor in the printer settings.

### What did you expect to happen instead?

The docs should look like this:

----

In PrusaSlicer or SuperSlicer you have to chose *Firmware-specific* in:

    Print Settings > Output options > Output file > Label objects

[image]

And you maust ensure that in the printer settings you choose *Klipper* as G-code flavor. You’ll find this in:

    Printer Settings > General > Firmware > G-code flavor

[image]

### How to reproduce this bug?

read the docs ;-)

tweh, 2023-11-25:
Found the doc report and re-postet the issue there.

closed 2023-11-25 by tweh

meteyou, 2023-11-25:
@tweh sry... i don't find your re-post...

tweh, 2023-11-26:
I’m sorry! Could have linked it …

Her it is: https://github.com/mainsail-crew/docs/issues/41

meteyou, 2023-11-26:
ohh ok. this is the old repo. the new docs are in https://github.com/mainsail-crew/gb-docs