mainsail-crew/mainsail #1668: Webcam not working when using port other than port 80.

delicado, 2023-12-02:
### Mainsail Version:

2.6.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

I tried accessing klipper via google VPS. Using SSH tunneling on my host machine, I could not use port 80. So I use 8080 instead, but I could not access the webcam. I fixed it by modifying the Mainsail installation of kiauh. The issue was in /assets/b7111f84.js on my installation of Mainsail. It's already minified, but I hope you can locate the exact source file. The code looks like this:
```js
getters$n = {
    getUrl: (o) => "//" + o.hostname + (o.port !== 80 ? ":" + o.port : ""),
    getHostUrl: (o) => (o.protocol === "wss" ? "https" : "http") + "://" + o.hostname + "/",
    getWebsocketUrl: (o, l) => o.protocol + ":" + l.getUrl + "/websocket",
  },
```
Changing it to fixed the problem:
```js
getters$n = {
    getUrl: (o) => "//" + o.hostname + (o.port !== 80 ? ":" + o.port : ""),
    getHostUrl: (o) => (o.protocol === "wss" ? "https" : "http") + "://" + o.hostname + ":" + o.port + "/",
    getWebsocketUrl: (o, l) => o.protocol + ":" + l.getUrl + "/websocket",
  },
```

### What did you expect to happen instead?

It should work.

### How to reproduce this bug?

Make a proxy for your mainsail. I did it with ssh tunneling to other computer, in my case it is GCP VPS.
`ssh -R 8080:localhost:80 -N username@host_address`

meteyou, 2023-12-02:
you have to use an absolute path in your webcam settings. changing the `getUrl` function will create many other issues.

in an older version we changed this behavior some time before, but there was multiple other issues with cross installations. so i don't think i will/can fix this with creating multiple other issues.

closed 2023-12-02 by meteyou

delicado, 2023-12-03:
Oh so that's what the textbox for on the Webcam settings. I see, thanks.

meteyou, 2023-12-16:
@delicado i think i found a fix for this issue. pls check the next release.

PR with the fix: https://github.com/mainsail-crew/mainsail/pull/1566