mainsail-crew/mainsail #1675: Spool ID not displayed when selecting spool

Ohara2208, 2023-12-07:
### Mainsail Version:

v2.8.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When changing spool on web interface the Spool ID is not displayed.

### What did you expect to happen instead?

Spool ID to be Displayed

### How to reproduce this bug?

Change spool on the web interface and when the popup displaying your rolls comes up you will see the lack of Spool ID

### Additional information:

[image]

meteyou, 2023-12-07:
this is already fixed: https://github.com/mainsail-crew/mainsail/pull/1605

Ohara2208, 2023-12-08:
> this is already fixed: #1605

Has this been rolled out? Im on V2.8.0 and it still doesnt show. 
Spoolman 0.14.0

[image]

meteyou, 2023-12-08:
No sry... Next release will be in the next days.