mainsail-crew/mainsail #1676: add info box on bed mesh

Stooovie, 2023-12-08:
### Requested feature:

I propose adding an info box in the bed mesh UI that would state that slight unevenness is normal and is normally taken care of by ABL.

### Solves the following problem:

3D Printing subreddits are full every day of users worried about their bed mesh, even when their range is under, say, 0.3mm. People just expect perfectly planar bed and freak out for no reason when they do not get it.

dw-0, 2023-12-08:
Imho that't not a task for a Webinterface like Mainsail. Such thing belong in some wikis, guides or whatever, but not in some info boxes throughout the interface. Mainsail would be full of info boxes if we put in explanations for common stuff users worry about. That's something every new user has to learn, read and understand. I don't think that this "education" should be done by Mainsail.

Just my 2 cents....

DustSwiffer, 2024-01-11:
I dont think people which recently joined 3d printing are going to install klipper right away. mostly they are  somewhat experienced already and know that small unevenness is not going to affect that much. In that case it makes more sense to spread this information in documentation of printer manufacturers as they are more likely to read that before diving deeper in more advanced stuff.

meteyou, 2024-01-12:
Some users want a print bed with minimal variance, and some users print with bed_mesh on beds, which I would not have thought is possible.
No matter what we write in this info box, we will annoy the users of one group with it.

Furthermore, I do not think that Mainsail should provide "general" information and should only report on the current state of the firmware. Therefore, I will close this feature request now. Thanks again for your input/idea.

closed 2024-01-12 by meteyou