mainsail-crew/mainsail #1699: Unable to click on object in "Exclude Object" modal

NV4RE, 2023-12-19:
### Mainsail Version:

v2.8.0

### Browser:

Other (Please tell us below)

### Device:

Other (Please tell us below)

### Operating System:

Other (Please tell us below)

### What happened?

In the `Exclude Object` modal, I am unable to click on the preview window to exclude the object I want. However, clicking on the list item works fine. I also noticed that when hovering over the preview window, it blinks rapidly. I'm not sure if it's a Vue-related issue or something else, but it seems like some open source projects encounter similar problems when the page updates rapidly, causing events not to register with the element in time.

### What did you expect to happen instead?

I should able to click on preview window to exclude object.

### How to reproduce this bug?

1. Open Mainsail web interface.
2. Print something.
3. Click the [x] (exclude object icon).
4. Attempt to click on the object you want to exclude.
5. Sometimes, the mouse event does not register.

### Additional information:

https://github.com/mainsail-crew/mainsail/assets/13000056/19323593-3932-4678-87f8-0b62cdf47087

Browser
- Brave
- Chrome
- Safari

Device:
- Desktop
- Laptop
- Tablet

Operating System:
- MacOS (Sonoma 14.1.1)
- iPadOS (17.3)
- Windows (11)
- ZorinOS

meteyou, 2023-12-19:
pls upload this gcode file. did you use PS 2.7 to create this gcode file?

NV4RE, 2023-12-19:
> pls upload this gcode file. did you use PS 2.7 to create this gcode file?

I use OrcaSlicer 1.8.1

NV4RE, 2023-12-19:
[gcodes-20231119-135243.zip](https://github.com/mainsail-crew/mainsail/files/13712144/gcodes-20231119-135243.zip)

PR #1719 (2023-12-30): fix(exclude objects): fix tooltip position in object map
## Description

This PR fix the position of the tooltip in the exclude objects map.

## Related Tickets & Documents

fixes #1699 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2023-12-30 by meteyou