mainsail-crew/mainsail #1701: Webcam aspect ratio incorrect when switching cameras

adamcstorm, 2023-12-20:
### Mainsail Version:

2.9.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When switching between two cameras with different aspect ratios, the aspect ratio doesn’t update so one cam looks stretched and the other looks squished if I go directly to said camera from the other. This happens on Chrome in Windows and Safari on iOS.

### What did you expect to happen instead?

When switching from one camera to the other, the aspect ratio of the active camera should update

### How to reproduce this bug?

On the webcam panel of the dashboard, or on the dedicated webcam page, select a single camera, then select a camera with a different aspect ratio. If the "all" selection is chosen, the aspect ratios will reset. You can then select any camera individually and the aspect ratio will be correct.

### Additional information:

I have two webcams: a pi cam v3 and a cheap AliExpress endoscope I use as a nozzle cam. Both are set up and functioning well. The cheap endoscope is a different aspect ratio than the pi cam. When switching between the cameras, the aspect ratio doesn’t update so the pi cam looks stretched and the endoscope looks squished if I go directly to said camera from the other. If I switch the view back to all camera, the aspect ratios return to normal and I can select an individual camera without issue. Pictures are attached but might not be obvious at first. I believe this is a bug. Perhaps there’s a setting I’m missing that can mitigate this issue?

Example images below.

crowsnest.cfg:

[crowsnest]
log_path: /home/astorm/printer_data/logs/crowsnest.log
log_level: verbose                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: true

[cam 1]
mode: camera-streamer                        # ustreamer - Provides mjpg and snapshots. (All devices)
                                        # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                      # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                         # Set different ports for each device!
port: 8080                              # HTTP/MJPG Stream/Snapshot Port
device: /base/soc/i2c0mux/i2c@1/imx708@1a                     # See Log for available ...
resolution: 4608x2592
custom_flags: --camera-width=4608 --camera-video.height=1080 --camera-snapshot.height=1080 --camera-stream.height=1080 -camera-options=AfMode=2 -camera-options=AfRange=1
#resolution: 1920x1080                     # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

[cam 2]
mode: ustreamer                        # ustreamer - Provides mjpg and snapshots. (All devices)
                                        # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                      # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8555                         # Set different ports for each device!
port: 8081                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/v4l/by-id/usb-Generic_HD_camera-video-index0                     # See Log for available ...
resolution: 1280x720
#custom_flags: --camera-video.height=720 --camera-snapshot.height=720 --camera-stream.height=720
#resolution: 1920x1080                     # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.

[image]
[image]
[image]
[image]
[image]

meteyou, 2023-12-23:
pls post your webcam settings in mainsail. I cannot reproduce this issue...

adamcstorm, 2023-12-23:
Here are the settings for both cameras. If it is a settings mistake that I made, I apologize. I appreciate you looking into the issue. 

[image]
[image]

PR #1707 (2023-12-26): fix: fix aspectRatio in MjpegstreamerAdaptive
## Description

This PR fix the aspectRatio, when you switch between different MjpegstreamerAdaptive webcams.

## Related Tickets & Documents

fixes #1701 

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2023-12-26 by meteyou