mainsail-crew/mainsail #1702: Additional filtering on Spoolman integration

adamcstorm, 2023-12-20:
### Requested feature:

I'm loving spoolman and the Mainsail integration! 
I have some suggestions for the spool selection dialog:

- Add filter ability for filtering for filament type, color, brand, etc.
- In the filament selection on print start, automatically apply a filter for filament type. There's already a helpful indication when the filament type doesn't match the gcode. I would hope that this could be enabled/disabled in the settings somehow as I expect this would not be desired in all cases.

### Solves the following problem:

When there are a large number of spools, it can be cumbersome to find the right spool. This would help speed up the process, particularly on mobile/small screens.

meteyou, 2023-12-21:
You can use the current search bar to filter everything. So filament type, color, and brand can all be filtered already.

adamcstorm, 2023-12-21:
That is a handy feature. However, I am of the opinion that having a dropdown+checkbox for a filter is more useful, particularly on a touch interface. Having the ability for a filter to be automatically applied based on filament detected in gcode file would be super useful IMO. It's certainly not a deal breaker. Just something I thought could improve usability. There's already a sort-by option, and I suppose there's a path to extend that functionality to a filtered list.

rackrick, 2023-12-21:
Presetting the search with the filament typ if a wrong filament is deteced might be an option. It comes with the downside does names containing the keyword also will be shown but this is proably an edge case. Anyways... I think additional dropdowns or checkboxes will overload the interface. If we have to hide them behind a search menu or something you also could just type the name into the searchbar even on mobile devices.

meteyou, 2023-12-21:
I also wanted to add filament type. This will be implemented relatively quickly.

In terms of the type of implementation, the manufacturer should also be the same and fast. But I think it's the same as @rackrick. It could become too much with too little added benefit.

I will add the filament type as a dropdown and check the UI...

adamcstorm, 2023-12-21:
I trust you all with the UI. It’s been a pleasure to use! I appreciate you considering the request and look forward to seeing what you come up with if it’s deemed an improvement.