mainsail-crew/mainsail #1709: Console error and unexpected hover graph behavior

blonestar, 2023-12-23:
### Mainsail Version:

2.9.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

On graph hover it throws console error and it does not show box with temperatures.
[image]

### What did you expect to happen instead?

It should show box with information's on graph hover.

### How to reproduce this bug?

Just update Mainsail to 2.9.0
2.8.0 works fine, I was downgrade it to check.

blonestar, 2023-12-23:
I guess it is a duplicate of
https://github.com/mainsail-crew/mainsail/issues/1655

meteyou, 2023-12-24:
Thx for posting this issue. This bug has nothing todo with #1655. This bug comes with the light mode PR: #1580 .

i Fixed this PR already here: #1706 . I hope i can merge it as soon as possible and release a bugfix release.

closed 2023-12-25 by meteyou