mainsail-crew/mainsail #1710: Webcomes are not visible in second printer

RBSECW, 2023-12-23:
### Mainsail Version:

2.9.0

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Two klippers are set up on the same PI, crownsnest is set for first printer. 

Mainsail for the first printer can show both cams, mainsail for the second printer can not show webcams. Shows FPS and black window instead of camera feeds. No error given 

Screenshots are showing Mainsail showing cams only for the first printed and Fluidd in both. Mobilraker and Octoeverywhere do not have problems either
[image]
[image]

### What did you expect to happen instead?

Mainsail to be able to show cams in both printer interfaces

### How to reproduce this bug?

2 (or more) klipper instances and at least 1 webcam

meteyou, 2023-12-24:
Pls Post the settings... Without settings, nobody can help you...

RBSECW, 2023-12-24:
Its not a config problem, as config in printer 2 is the same as in printer 1. Screenshot from FluidD shows that this config works in FluidD. It also works in mobilraker and octoeverywhere. I do not need help with Mainsail, rather than bug being fixed. Same issue reported several time over the net last couple of days.

meteyou, 2023-12-24:
There are different webcam settings in Mainsail. To fix a bug, I need to be able to reproduce it. Without your config, how you have it configured in your second printer is a guess. If you don't want to help me and don't want to post your config, I'd rather close this post. Thanks for reporting it, but I cannot fix a bug that way.

RBSECW, 2023-12-24:
Wow, thats a good way of getting rid of a bug - just ignore it =) Do you need config of crownsnest? Mainsail? webcam in mainsail?

meteyou, 2023-12-24:
If this issue really have more people, maybe an other user will also post it... I didn't see it until now, but feel free to add links to there issues, that i can read all of that (I don't have the time to read every single communication platform with 3d printer topics on this world...)

because we are speaking about a Mainsail issue, i would need at first the Mainsail config (as i wrote above). The Crowsnest config could be also helpful, but its not really necessary in this specific case. (to generic... the more information we get, the better we can work. unfortunately, none of us can do anything with 2 screenshots...)

PR #1713 (2023-12-26): fix: fix webcam url with multiple moonraker instances
## Description

This PR change the hardcoded Moonraker ports to automatic read ports from the Moonraker config.

## Related Tickets & Documents

maybe fixes #1710 but without any config, I don't know if this is the issue...

## Mobile & Desktop Screenshots/Recordings

none

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2023-12-26 by meteyou