mainsail-crew/mainsail #1717: JS exception on mouseover temperature graph

mrThe, 2023-12-26:
### Mainsail Version:

v2.9.0

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

macOS

### What happened?

Graphing does not show temperature on mouseover and rises an error instead:
```
Uncaught ReferenceError: $vuetify is not defined
    tooltipFormatter index-2d058d3f.js:28946
    _showTooltipContent echarts-9bc570b0.js:22
    _showAxisTooltip echarts-9bc570b0.js:22
    _showOrMove echarts-9bc570b0.js:22
    _showAxisTooltip echarts-9bc570b0.js:22
    _tryShow echarts-9bc570b0.js:22
    manuallyShowTip echarts-9bc570b0.js:22
    b echarts-9bc570b0.js:22
    co echarts-9bc570b0.js:22
    eachComponent echarts-9bc570b0.js:22
    co echarts-9bc570b0.js:22
    du echarts-9bc570b0.js:22
    A echarts-9bc570b0.js:14
    du echarts-9bc570b0.js:22
    dispatchAction echarts-9bc570b0.js:22
    KP echarts-9bc570b0.js:22
    e echarts-9bc570b0.js:22
    trigger echarts-9bc570b0.js:14
    dispatchToElement echarts-9bc570b0.js:14
    mousemove echarts-9bc570b0.js:14
    trigger echarts-9bc570b0.js:14
```
[image]

### What did you expect to happen instead?

Tooltip with temperature should showup

### How to reproduce this bug?

Drag mouse across the graph

### Additional information:

Mainsail is installed using kiauh and updated a few times since. I don't know when exactly it starts to error, but it reproducible on any browser i have: desktop/mobile safari/chrome/firefox.

I assume i need to rebuild the assets, but i have no idea how.

rackrick, 2023-12-26:
Thanks for reporting.
That's an known issue and already fixed https://github.com/mainsail-crew/mainsail/pull/1706

mrThe, 2023-12-26:
@rackrick thanks for the update. Closing.

closed 2023-12-26 by mrThe

meteyou, 2023-12-26:
I hope we can publish the Bugfix release on the next days. Thx for the report.