mainsail-crew/mainsail #1730: G-Code tab suggestions ignore primary color setting

wlhlm, 2024-01-04:
### Mainsail Version:

2.9.1

### Browser:

Firefox

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

G-Code tab suggestions appear blue in terminal instead of red which is my primary color in the Mainsail UI settings.

[image]

### What did you expect to happen instead?

G-Codes suggested when hitting tab to complete the current command line should appear in primary color in the terminal.

### How to reproduce this bug?

Go to the G-Code console, type `SET_`, hit the _tab_ key, Mainsail then suggests a number of G-Codes with the `SET_` prefix for completion.

meteyou, 2024-01-04:
Yes. This is hardcoded to blue and have nothing Todo with primary color.

wlhlm, 2024-01-04:
I see. Though I don't understand why it would need to be hardcoded and ignore the UI color setting. The tab suggestions seem to work the same way as for example G-Codes put into the console through other means (such as pressing movement buttons), namely clicking on them puts them console input field.

My workaround for now is the following snippet in my theme CSS:
```css
/* See https://github.com/mainsail-crew/mainsail/issues/1730 */
a.command.blue--text {
    color: var(--v-anchor-base) !important;
}
```

PR #1733 (2024-01-29): fix(console): fix color of autocomplete and command list
## Description

This PR will switch the color of the Command List or Console autocomplete from blue to primary.

## Related Tickets & Documents

fixes: #1730 

## Mobile & Desktop Screenshots/Recordings

before:
[image]
[image]

after:
[image]
[image]

## [optional] Are there any post-deployment tasks we need to perform?

none
closed 2024-01-29 by meteyou