mainsail-crew/mainsail #1731: Direct link to specific printer in multi-printer setup

kondas, 2024-01-04:
### Requested feature:

The possibility to  have a direct link to specific printer when multiple printers are handled by a single Mainsail instance

### Solves the following problem:

OrcaSlicer can send print jobs directly ro Moonraker API, but to have device control only an embedded web browser is available. 
It would be nice to have direct link to a specific printer in Mainsail for this purpose, because currently the printer has to be selected after switching to device tab.

### Additional information:

Happy new year!

meteyou, 2024-01-05:
thanks for your request. something like this has been asked for several times, but i don't think it will be able with every variant. it might be most useable with the json instances store. how have you currently configured your multi instances?

kondas, 2024-01-05:
I'm using the json store. Is it possible already? I've tried searching to no avail

kondas, 2024-01-05:
Thanks for clarification. 

If there is possibility to modify something in the HTTP request from the client would be adequate, that way one could make each printer accessible under a separate hostname by ie. setting a header value in the reverse proxy

napter, 2024-01-11:
Is there still no way to add param to the url so mainsail starts with that printer? It would be a lot cleaner if we could create a bookmark to link to a specific printers dashboard.

meteyou, 2024-01-11:
@napter this is an open feature request.. so no. there is no integrated function for that.

meteyou, 2024-04-27:
this FR is fixed with the PR #1837

closed 2024-04-27 by meteyou