mainsail-crew/mainsail #1735: Add camera authorization.

khseal, 2024-01-07:
### Requested feature:

Many cameras require authorization.
Example http://admin:admin@192.168.5.100/image.jpg does not work.

meteyou, 2024-01-07:
URL authentification should work with the most webcams. I will not implement something webcam specific.

closed 2024-01-07 by meteyou

khseal, 2024-02-20:
It's not specific.
The link http://admin:admin@192.168.5.100/image.jpg opens in any browser.
mainsail cannot open. mainsail is specific and not a camera!

meteyou, 2024-02-20:
This is just a URL for a Snapshot. It needs a little more to display it correctly in Mainsail, not "only a link" but I would not see any issue as to why this should not work with MJPEG-ADAPTIVE or UV4L-MJPEG mode in Mainsail.

But in UV4L-MJPEG, you will only see the last image. Not a video.

We already implemented ten different webcam services for a web interface built to control your Klipper printer. This is no "webcam control center". So please use a webcam we support, or feel free to make a fork and implement your specific webcam mode.

khseal, 2024-02-20:
I have cameras with openipc firmware. https://openipc.org/
The telegram bot just works without any problems.
Mainsail fell off when authorization was added to the camera.

khseal, 2024-02-20:
Hm, my bot can send video.
[image]

meteyou, 2024-02-21:
@khseal, you send a screenshot from a video... Sorry. But all of this sounds more like a joke/scam... I will lock this post now because this is only a waste of time for me.