mainsail-crew/mainsail #1741: Second instance unable to use crowsnest or any other camera

kg6ebj, 2024-01-09:
### Mainsail Version:

V2.9.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When installing 2 printers, 1 host, and using web cameras, the first printer (printer_1_data) controls Crowsnest.
The second printer (printer_2_data) can add the cameras in the webcams settings, but all other references for webcams are not present and the webcam is not accessible to plugins like mainsail-timelapse or even to view in the dashboard.

### What did you expect to happen instead?

The cameras should be available in both instances. 

### How to reproduce this bug?

Clean install current mainsail
Install crowsnest
Add 2 instances of klipper/moonraker
Add at least 2 cameras
map the cameras in the printer session that is controlling crowsnest (works as expected)
map the cameras in the printer session that does not control crowsnest (shows preview in the Webcams control panel but is not usable anywhere else in mainsail.)

### Additional information:

This used to work; I can't remember when it stopped working because I had system issues and stopped running 3d prints for a bit.

I expect this is a function that print farms would need/require to run multiple cameras into 1 instance of Mainsail.

meteyou, 2024-01-09:
Double checked it right now and work like it should.

closed 2024-01-09 by meteyou

jgarbe, 2026-11-13:
I have two instances of Klipper and  Moonraker on my Klipper 3D print server.  Each of the instances have their own config file folder," printer_1_data", and "printer_2_data".  Crowsnest will only install a single instance.  I put the single instance -- "crowsnest.conf"-- in the "printer_1_data" file.  Make a symlink to the second instance with --
`
$ ln -s ~/printer_1_data/config/crowsnest.conf ~/printer_2_data/config/crowsnest.conf`

The second printer instance will now have the WebCam options.