mainsail-crew/mainsail #1752: Cannot See GCode in VIewer when generating in OrcaSlicer

TopoDiFogna, 2024-01-20:
### Mainsail Version:

v2.9.1

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Uploaded a gcode file and it is not visible in the GCode Viewer. Browser console logged 2 errors:
```
Missing feature Skirt
```
```TypeError: Cannot read properties of undefined (reading 'color')
    at t.value (Viewer-8db94310.js:1:1579265)
    at a.<anonymous> (Viewer-8db94310.js:1:1593681)
    at h (Viewer-8db94310.js:1:1539484)
    at Generator._invoke (Viewer-8db94310.js:1:1539230)
    at Generator.next (Viewer-8db94310.js:1:1539843)
    at jo (Viewer-8db94310.js:1:1544775)
    at r (Viewer-8db94310.js:1:1544972)
    at Viewer-8db94310.js:1:1545033
    at new Promise (<anonymous>)
    at a.<anonymous> (Viewer-8db94310.js:1:1544919)
 ```

### What did you expect to happen instead?

Preview the gcode file in the GCode Viewer

### How to reproduce this bug?

Generate some gcodefile with orcaslicer and try it in the GCode Viewer

### Additional information:

I saw that mainsail is using "@sindarius/gcodeviewer": "^3.2.2" to display gcodes files, using the latest version (3.7.6 when writing this), also available at https://sindarius.com/ does not have the problem.

closed 2024-01-26 by meteyou

Exeu, 2024-02-17:
Hi @meteyou,

i updated to the most recent mainsail version, and still get the issue with Missing Skirt.

[image]

TopoDiFogna, 2024-02-17:
That's odd, i tried with multiple gcodes locally on my fork and It was working.

I'll debug further later today

TopoDiFogna, 2024-02-17:
So these are screenshots taken now from my PC, both instances point to my server running moonraker.
Left is master, fresh cloned, served via `npm run start`, right is my remote server which i just updated:
[image]
[image]

Seems that `Missing feature Skirt` is not an issue since it is displayed also on my local.

More worrying are the 2 `TypeError`s displayed on the remote which at this point are the culprit of everything. But I cannot reproduce them locally 😢 

I see 2 mayor differences between the instances:

1. the one served locally is running on service worker but the other no, since that domain is just internal to my network so no need to serve it via https
2. The hashes of the Viewer.js differ but the source is the same. Maybe i'm build with a different version of node (Node v20.10.0 and npm 10.2.3) which may produce different results

TopoDiFogna, 2024-02-17:
Ok I managed to make it work on the remote also.

Seems a bug in the library used, @Exeu can you please try to change the ColorMode inside the viewer to Extruder for example? It should work.

Seems that launching in local even if ColorMode is set to Feature is still using Extruder, that's why I didn't notice it in the first place. (@meteyou maybe there is a bug here?)

Anyway using the Viewer @ https://sindarius.com/ the issue is reproducible, so it's not a bug of mainsail but of the library itself

Exeu, 2024-02-17:
@TopoDiFogna for the other types / views it ist working. But not for feature.

Exeu, 2024-02-17:
@[image]

Here is the gcode I am using