mainsail-crew/mainsail #1753: Lagging after last update

DVSVIDEO, 2024-01-20:
### Mainsail Version:

v2.9.1

### Browser:

Other (Please tell us below)

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

Today I updated one with the automatic update, which updates all components.
After the update the clipper would not start because the firmware of the EBB 2209 CAN card was old for it.
I put new firmware on it (both on the sub board and on the rpi) (you can see it in the attached pictures)
After that it started, but the startup was slower than usual and there is a drastic slowdown during operation.
The webcam refreshes with 00-01 frames instead of the usual 18-22 frames and the system often loses connection and reconnects with "initialize".
I suspect that the connection is faulty because the systems load is low below 6%.
[image]
[image]

### What did you expect to happen instead?

I expect the old familiar speed of operation

### How to reproduce this bug?

I simply upgraded the raspberry system, then the CAN adapter (BTT 2209) needed a firmware update.

### Additional information:

My browser: Brawe (chrome based)

meteyou, 2024-01-20:
klipper updates have nothing todo with mainsail in this area... mainsail just communicate with moonraker.

DVSVIDEO, 2024-01-20:
The mistake was mine. Rebooting my router fixed it. I apologize

closed 2024-01-20 by DVSVIDEO

meteyou, 2024-01-20:
@DVSVIDEO thx for the feedback! happy printing!

DVSVIDEO, 2024-01-20:
Thank you for your excellent work!
I really like Mainsail.