mainsail-crew/mainsail #1756: [Spoolman] Scan QR code from Mainsail Spoolman snippet

ramaxa9, 2024-01-26:
### Requested feature:

it would be very usefull to have this feature
and even better, to execute custom g-code when spool is selected, for example to pre-heat extruder and load filament, and maybe pass spool id to g-code for future ideas

meteyou, 2024-01-26:
i think its not a part of the web interface to scan a qr-code..

and the next thing is, that mainsail is only a client. so executing automatic gcode would be the wrong place in mainsail. for example. if you don't have the web interface running and select a spool on your klipperscreen display, mainsail is not running and cannot execute any gocde. so i think this feature should be located in the moonraker spoolman plugin, or in spoolman itself (but then, its not printer specific).

ramaxa9, 2024-01-26:
ok i can buy 2d barcode scanner, but there is no serch results when i enter spool id in the search field. Maybe an field to enter spool id?

m-klecka, 2024-02-12:
Just install mobileraker on your phone. It has spoolman integration now and can scan the qr codes and select them "for mainsail"

JekleFPV, 2024-08-01:
Forgive me for asking a n00b question here, but this is still just in the development branch, right?

I very much prefer Mainsail over Fluid, but the latter does have a button clear and present to get it to scan. Would love to have the same in mainsail, and to have it show up in klipperscreen, so I could just tap that whenever I decide to change a spool.

meteyou, 2024-08-01:
@JekleFPV No, this is already published, but this is only a function to search for the QR code in the search input field—nothing more. So, if you use a barcode scanner, you can use this.

If you want a button in Klipperscreen or on your printer, this has to be implemented somewhere else. As I described above, Mainsail is just a web interface, and "scanning a webcam image" is not a good solution, so I will not implement the same function as Fluidd for this.