mainsail-crew/mainsail #1758: Z Offset is not saved/applied correctly in printer.cfg

radry, 2024-01-28:
### Mainsail Version:

v2.9.1

### Browser:

Firefox

### Device:

Mobile device (Phone, Tablet or similar)

### Operating System:

Android

### What happened?

Changed the Z offset with the UI elements either before or during a print and press "save". Then after the print finished or is canceled press save config on top of the screen (as instructed) to apply the new Z offset to the printer.cfg and to restart klipper.  

After the restart the new Z offset is not applied but instead reset to 0. Starting a print again confirms this as the offset is clearly wrong during the first layer. 

When looking in the (new) printer.cfg it just automatically commented out "position_endstop" of the Z axis but didn't apply a new one.

### What did you expect to happen instead?

Adjust position_endstop of the Z stepper in printer.cfg to the new value.

### How to reproduce this bug?

1) Change the Z offset with the UI elements either before or during a print and press "save".
2) (After the print finished or is canceled) Press "save config" on top of the screen (as instructed) to apply the new Z offset to the printer.cfg
3) Wait for restart 
4) Check Z offset by either starting the print again, paper method and observe that the Z Offset is wrong.
**5) Look at printer.cfg** -> no new position_endstop or other way of offest was applied.

### Additional information:

I use manual bed leveling. No Probe, just endstop.

meteyou, 2024-01-28:
Thanks for your bug report, but Mainsail only send the G-Code to Kipper. The calculation and config editing happens only in Klipper. So this could only be a Klipper issue.

closed 2024-01-28 by meteyou

radry, 2024-01-28:
Are you sure it sends the correct gcode though?

meteyou, 2024-01-28:
you dont upload a log or a screenshot from your console, but i think so, because you wrote, that it comment out the original offset:
> When looking in the (new) printer.cfg it just automatically commented out "position_endstop" of the Z axis but didn't apply a new one.
but i think more it is an issue with your config file itself. klipper is very strict with the SAVE_CONFIG part in the config. if you just need help, feel free to use our discord or our discussions area.

TMTYD, 2024-11-22:
If you correct the z-offset during printing, it will be saved incorrectly or not at all.

If you go to + 0.005 for the z-offset, then this value is not added but subtracted. And if you go to -0.005, then this is added.

Assuming you have a Z-offset on the end stop of 4.220, if you go to +0.005, it becomes 4.215, if you go to -0.005, it becomes 4.225

In other words, it leads to confusion if you adjust the z-offset during printing.

In addition, if you have set the z-offset and you click Save Config after printing, this value is not saved in the printer.cfg.

You have to click on Save in the Z-Offset window or enter this command in the console,
Z_OFFSET_APPLY_ENDSTOP: Adjust the z endstop_position

then the Z-offset will also be saved in the Printer.cfg

[image]
[image]
[image]
[image]

meteyou, 2024-11-22:
> If you go to + 0.005 for the z-offset, then this value is not added but subtracted. And if you go to -0.005, then this is added.
Assuming you have a Z-offset on the end stop of 4.220, if you go to +0.005, it becomes 4.215, if you go to -0.005, it becomes 4.225

Thats right. you dont change the "hardware offset", because you just change the "gcode offset" and if you add your "gcode offset" to the "hardware offset", it looks wrong at the first moment, but its right. A higher endstop position is a lower nozzle.

> In addition, if you have set the z-offset and you click Save Config after printing, this value is not saved in the printer.cfg.
You have to click on Save in the Z-Offset window or enter this command in the console,
Z_OFFSET_APPLY_ENDSTOP: Adjust the z endstop_position
then the Z-offset will also be saved in the Printer.cfg

Thats normal, because you have to click on the "save" button in the offset panel (it just execute the `Z_OFFSET_APPLY_ENDSTOP` gcode, nothing more) and then you have to click on SAVE_CONFIG (or just use the button in the overlay to execute the SAVE_CONFIG command). if you just hit the SAVE_CONFIG button without clicking or typing the `Z_OFFSET_APPLY_ENDSTOP` command, it will not change the offset in your config. maybe you use a bed_mesh or something else, that the SAVE_CONFIG button is active, but its not the offset.

TMTYD, 2024-11-23:
Ok thanks for the answer

silentjay1980, 2025-04-26:
ok sorry to hijack but this thread is linked in a lot of places so maybe someone has a similar issue to me. my save button is always greyed out i cant use it ever. i have a k1c and i manually hi -.050 everytime i started a print because that lowers the nozzle for an acceptable layer for me, id like to make it permanenet but i never see a no greyed out save button.

[image]

[image]

meteyou, 2025-04-26:
@silentjay1980 if you need help, pls don't hijack other issues, when it is not exactly the same issue you have. for help itself, pls join our discord server or use the discussion section (the second screenshot itself is fluidd, there for, we cannot help anyway). here is the bot with the "help instruction" already: https://github.com/mainsail-crew/mainsail/issues/1758#issuecomment-1913590778