mainsail-crew/mainsail #1774: [MacOS]: Websocket infinite reconnection for locally hosted instance

wildalc, 2024-02-12:
### Mainsail Version:

v0.8.0-310

### Browser:

Chrome

### Device:

Laptop

### Operating System:

macOS

### What happened?

The issue was initially posted here https://github.com/mainsail-crew/mainsail/issues/1632, https://github.com/Arksine/moonraker/issues/726 and https://github.com/Rat-OS/RatOS/issues/106.

Similar issue was reported on [Reddit](https://www.reddit.com/r/klippers/comments/16u2jt9/cant_connect_to_moonraker_ip_on_mac_but_can_on/?utm_source=share&utm_medium=web2x&context=3)

### What did you expect to happen instead?

It should work on MacOS Monterey, Ventura, Sonoma.

### How to reproduce this bug?

1. Follow RatRig's installation instructions https://os.ratrig.com/docs/introduction/
2. After the setup, open http://ratos.local/ or http://<IP_ADDRESS>.
3. Open the networks tab, and see the WebSocket infinite reconnection request
4. Only port **80** does this, but when you visit **port 7125**, or **ratos.local/configure** it works perfectly.
NOTE: the same problem was reproduced with [Voron Installing Mainsail instructions](https://docs.vorondesign.com/build/software/installing_mainsail.html)

```
**cat ../logs/moonraker.log**

2024-02-12 19:10:49,889 [application.py:log_request()] - 101 GET /websocket (192.168.10.152) [_TRUSTED_USER_] 4.53ms
2024-02-12 19:10:49,890 [websockets.py:open()] - Websocket Opened: ID: 3898713448, Proxied: True, User Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/121.0.0.0 Safari/537.36, Host Name: 192.168.10.151
2024-02-12 19:10:49,896 [websockets.py:on_close()] - Websocket Closed: ID: 3898713448 Close Code: None, Close Reason: None, Pong Time Elapsed: 0.01

**cat mainsail-access.log**

192.168.10.152 - - [12/Feb/2024:20:02:11 +0000] "GET /websocket HTTP/1.1" 101 0 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/121.0.0
.0 Safari/537.36"
192.168.10.152 - - [12/Feb/2024:20:02:13 +0000] "GET /websocket HTTP/1.1" 101 0 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/121.0.0
.0 Safari/537.36"
192.168.10.152 - - [12/Feb/2024:20:02:15 +0000] "GET /websocket HTTP/1.1" 101 0 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/121.0.0
.0 Safari/537.36"

**cat mainsail-error.log**
<empty>
'''

### Additional information:

The problem is see with MacOS Monterey, Ventura, Sonoma (running on Intel, 2018).
I tried older laptops with versions MacOS BigSur, Catalina and it works fine (running on Intel, 2012).
On my new M1 with fresh MacOS Sonoma it works.
MacBook-2018 was upgraded from Ventura to Sonoma with no luck. Please note that on this machine I played with Proxy settings before upgrade.

In 2022 I has similar issue, but then I was able to add printer's IP address with port 7125 under **Interface Settings** -> **Printers** -> **ADD PRINTER**. I had to do that every time I started the 3D printer (running RatOS). Now when I check this settings on my working MacBook, I see the IP address with port 7125, which was added automatically.

meteyou, 2024-02-13:
Thanks for this bug report, but did you read the last message in the old bug report? https://github.com/mainsail-crew/mainsail/issues/1632#issuecomment-1847934051
please don't open new issues for the same issue...

Furthermore, there has never been a Mainsail version with the version number v0.8.0-310...

But if it works for you, even on different devices with the same operating system, it sounds very much like a problem with your device (wrong network/proxy settings).

Port 80 successfully opens the web interface (you see Mainsail itself), but it cannot connect to your Moonraker instance (the connecting/init dialog). If you open <IP>:7125 or ratos/configurator, you only open an HTTP web interface. These don't connect with the WebSocket. Your issue is only the WebSocket connection!

But as I wrote already in the other issue: as long as I cannot reproduce this issue, I'm not able to fix it...

meteyou, 2024-02-13:
I did not say that the OS generally causes issues. I have been working exclusively on Mac for a long time and never had this problem. It's a setting of your specific setup on the 2018 Intel device...

a reverse proxy is always a special setup. we do not officially support it, so you are on your own. but reverse proxies need their own settings for a WebSocket connection, as this socket must be kept open. but this is generally a reverse proxy "issue".

wildalc, 2024-02-13:
Thanks for your reply. On MacBookPro there is yet a bridge and virtual interfaces, which can mix up the source interface/address when reaching a webserver (in our case the reverse proxy). So before mirroring the switch port where my printer resides and wireshark the communication, I decided to delete all virtual interfaces. Rebooted the system and that fixed the problem.

So here is step by step what helped me:
1. On your Mac, choose `Apple menu`  > `System Settings`, then click `Network`  in the sidebar.
2. On the right, scroll to the bottom of the window, click the Action pop-up menu `...v` , then choose `Manage Virtual Interfaces`.
3. Delete everything you have in there.
4. Reboot your MacBook and printer.

meteyou, 2024-08-17:
I think we found the issue (thanks to a user on Discord). It appears that the 'Screen Time' feature in macOS is causing the issue. If I set the Screen Time function on a website like 'reddit.com' (which was the exact case from the discord user), I can reproduce the issue.

Please check if you are using this feature in macOS. I would appreciate feedback to see if this is indeed the problem for everyone here. (After disabling it, a reboot may be necessary).

You can find the function here:
> screen time -> content & privacy -> store, web, siri & game center control -> access to web control set to "limit adult websites" -> customize

dderg, 2024-08-17:
I would like to add that after disabling the website restriction, you have to reboot the computer, otherwise the issue with infinite websocket reconnection persists.

jpwolfe9912, 2024-08-22:
@meteyou Holy crap! This worked! I've been dealing with this for the past couple years and have been beyond confused. Thank you for the help!

jinthagerman, 2025-02-08:
This worked for me too. You can however just set `Web Content` to `Unrestricted` instead of disabling all of "Content & Privacy".