mainsail-crew/mainsail #1782: 4th preheating preset replaces 3rd one

ajenkuhnabat, 2024-02-16:
### Mainsail Version:

0.2.9

### Browser:

Chrome

### Device:

Desktop PC

### Operating System:

Windows

### What happened?

When I try to add a 4th preheating preset instead of a new one beeing created it overwrites the data entered for the 4th entry in the 3rd entry, destroying it's contents irreversabely. That should not happen. If a limit exists I'd expect a message and prior information for that. 

### What did you expect to happen instead?

A 4th preset should be added.

### How to reproduce this bug?

Try to add a 4th preset

meteyou, 2024-02-16:
sry i cannot reproduce this issue:
[image]

but there is also an issue with your version number. v0.2.9 is maybe very very very old or just a typo. i already fixed some issues short before with the presets (i think it was with v2.9.1). but right now v2.10.0 is the newest version of mainsail. so please update your system and test it again.

closed 2024-02-16 by meteyou

ajenkuhnabat, 2024-02-20:
thank you. Of course you are correct. My actual version is 2.9.1. I'll update and check it out. Thank you very much for your efforts :)